AccelerationConsortium/ac-dev-lab #158: Custom nozzle camera for Bambu Lab A1 Mini

sgbaird, 2025-01-20:
@Neil-YL thoughts on mounting the camera to the toolhead similar to what's shown in [makerworld](https://makerworld.com/en/models/572839?from=search#profileId-493081) as a "nozzle cam"?

**Option 1**

[image]

[image]

**Option 2**
[Similar version](https://makerworld.com/en/models/820459)

[image]

**Option 3**
[Another one](https://makerworld.com/en/models/1009661#profileId-989002)

[image]

In our case, we'd want to image the print directly above, and we'd move the toolhead to get the camera in the right place.

This is the kind of tripod camera mount we've typically been using: https://www.pishop.ca/product/pro-mini-camera-mount-for-raspberry-pi-zero/ (image below). However, the plate that the camera is attached to is "wiggly" relative to the plate the Pi Zero 2W is attached to.

[image]

I lean slightly towards the Option 1 style (use a flexure clamping mechanism onto the toolhead). I'd be OK with replacing the cover (seems more mechanically stable and EDIT: it's a pinch-and-remove mechanism). I lean towards reusing the same hardware (nuts/bolts) that come with the "Pro Mini Camera Mount". 

Related:
- https://ca.store.bambulab.com/products/print-head-front-cover-a1-series (EDIT: the print cover seems to be removeable without hardware)

[image]

So, how about designing a new front cover that simply has a single 3d printed threads jutting out to be attached with the regular tripod nut? Or this could be the arm extension.

[image]

sgbaird, 2025-01-20:
[image]

sgbaird, 2025-01-21:
[image]

[image]

[image]

[image]

Neil-YL, 2025-01-21:
You work so quickly!

I’ll test this part tomorrow.  I am also thinking we can replace one of the AMS feeding tubes with the cable for the camera.

sgbaird, 2025-01-21:
We'll see if it turns out 😅 thanks!

Neil-YL, 2025-01-21:
[image]

[image]
The finishing of Carbon Fibre material is quite coarse, I will print a PLA to see if it is better.

Neil-YL, 2025-01-21:
> For provenance, link to Fusion 360 file: https://a360.co/4jq5PUd

Nice you've attached the file, I don't know why but the project file saved on Bambu doesn't allow me to change the material

Neil-YL, 2025-01-21:
[image]

[image]

I will dismount the camera next to the robotic arm and to see how it fits to the nozzle cover

sgbaird, 2025-01-21:
Awesome! This looks great. I have a bunch of reusable zip ties in the closet cupboard in my office. Could you grab some and work on getting the power cord attached with the rest of the tubing in the way that allows free movement? Lmk if we need a longer power cable and what length.

Really glad you looked into the temperature. Last night when I was telling my spouse about it, she brought that up, and I realized I missed that key detail 😅 thanks for checking that one.

Don't worry about streaming for now. I'll ping @Jonathan-Woo about getting the camera set up.

Neil-YL, 2025-01-21:
btw I also fix the wiggling issue by filling the gap (with our printed 0.1mm thick square) between the camera mount and the pico.

Neil-YL, 2025-01-21:
The printer is printing I would take one after it finishes.

Here’s a video for you to check out first, looks quite stable during the print job

https://github.com/user-attachments/assets/a058e103-e7c3-4587-a774-167a25fa23b6

sgbaird, 2025-01-21:
Awesome! I guess another question: what's our new usable area on the build plate? I'm not sure how far "back" the gantry can go (i.e., if all corners of the build plate are accessible). Even if we lose about 20% of the build plate, I don't think it's a big issue, but we'll just need to keep that in mind for the workflow.

Neil-YL, 2025-01-21:
Answered by their chatbot:

Based on the information provided in the knowledge sources, I can answer your question about the usable area on the build plate of the A1 mini.

The A1 mini has a maximum print size of 180 x 180 x 180 mm³. However, the actual usable area isslightly smaller due to certain limitations. According to the [Print volume limitations](https://wiki.bambulab.com/en/knowledge-sharing/print-volume-limitations) document, Bambu Lab has implemented some restrictions on the movement of the tool head in certain areas to achieve specific functions such as cutter cutting and Z hop.

For the A1 mini specifically, there is a detection area that limits the printable space. The A1 mini detection area is (156, 152) -> (180, 180). This means that there is a small portion of the build plate that cannot be fully utilized for printing.

It's important to note that these limitations are implemented in Bambu Studio to prevent issues during printing. The lost print volume in the front left corner is 18x28mm² for the A1 mini.

To summarize, while the maximum build volume is 180 x 180 x 180 mm³, the actual usable area is slightly smaller due to these implemented restrictions. The exact usable area is not explicitly stated, but it would be the build volume minus the detection area and the lost print volume in the front left corner.

For the most accurate and up-to-date information on the usable area, I recommend checking the latest version of Bambu Studio or contacting Bambu Lab support directly.

I will measure our current 121 square printing dimensions on the plate later

sgbaird, 2025-01-21:
Awesome! That seems pretty reasonable. Let's go with that.

Neil-YL, 2025-01-21:
How I fix the wiggle gaps, for later design we can print a camera mount with a stable fit:

[image]

Not sure about the quality and how close the camera should go for Sissi later evaluation processing, but it is possible that we can still have access to all the area of the plate:

[image]

sgbaird, 2025-01-21:
Oh, nice! Hmm.. can we get the camera pointing a bit more perpendicular to the plate? The camera should be pretty close to the build plate (i.e., as close as possible without making the rest of the toolhead crash into the plate). It would only be imaging a single square at a time. We'll keep in mind for later, but for now let's ignore that inaccessible 10% of the build plate.

sgbaird, 2025-01-22:
@Neil-YL could you export the 3mf file (directly from Bambu Studio using File --> export to keep all settings) and attach that here? Then, I'll fix up the submission at https://makerworld.com/en/models/1019952#profileId-1001000

Neil-YL, 2025-01-22:
> [@Neil-YL](https://github.com/Neil-YL) could you export the 3mf file (directly from Bambu Studio using File --> export to keep all settings) and attach that here? Then, I'll fix up the submission at https://makerworld.com/en/models/1019952#profileId-1001000

[A1 Mini Cover with tripod screw v3 v1.3mf.zip](https://github.com/user-attachments/files/18510588/A1.Mini.Cover.with.tripod.screw.v3.v1.3mf.zip)

sgbaird, 2025-01-22:
Reusable zip ties for power cord and power cord

[image]

sgbaird, 2025-01-23:
https://makerworld.com/en/models/1019952#profileId-1001000

sgbaird, 2025-01-23:
We might need some external lighting for when the office lights go off. Is there a built-in light, and if so, do you think it's bright enough to give us OK image quality? (not worrying about difference between office lights on vs. off, just if we can get OK images when office lights are off)

sgbaird, 2025-01-23:
I came across [Bright Pi](https://www.pishop.ca/product/bright-pi-bright-white-and-ir-camera-light-for-raspberry-pi/) with some nice [quickstart instructions](https://learn.pi-supply.com/make/bright-pi-quickstart-faq/) which should work in terms of mounting onto the camera, but it requires soldering the LEDs and header pins on and might obstruct some of the camera view. An easy fix for potential obstruction would be to only solder the outer positions (since the LEDs seem to be individually addressable).

[image]

(cc @brendenpelkie you might be interested in Bright Pi in the context of the overhead camera tool for Science Jubilee - feel free to unsubscribe from notifications for this thread once you see the message).

Alternatives:
- [LISIPAROIWHT-01 Add-On Board, LED Camera Light Ring For Raspberry Pi, White](https://canada.newark.com/cyntech/lisiparoiwht-01/lisiparoi-white-led-camera-light/dp/31AC3472) (not individually addressable, but probably doesn't matter for our application, and it's just a GPIO input which is much simpler)
- ~~[Adafruit: Cool White LED Ring Light with USB Cable and On/Off Switch - 70mm Diameter 5V](https://www.adafruit.com/product/5137) seems like a nice, easy option in terms of on/off and voltage, but **it will get hot**.~~ (EDIT: **this will get too hot**)
- [Adafruit LED Ring Light - 76mm Diameter (9V)](https://www.adafruit.com/product/4433), but even with the heatsink this might get pretty hot if used continuously. Also not straightforward to mount.
- This [RGBW neopixel ring](https://www.adafruit.com/product/2852) might be a nice option, but again, mounting and heat management aren't straightforward.

---

There might be some obstruction of the laser cut piece the camera is attached to.

[image]

Ignoring the red circles, as a ballpark, it looks like the hooks at the 90 degree angle would interfere with the camera.

[image]

I lean a bit towards designing a 3D printed piece (as @Neil-YL suggested earlier) that mimics the existing camera mount, thus removing the hooks at the end. This could even be printed directly as part of the front cover, eliminating the need for a tripod screw and nut. We can still reuse the screws, nuts, and standoffs from the camera mount kit.

Ballparking, it looks like there might be enough clearance for the Bright Pi or the LISIPAROIWHT-01 relative to the rest of the toolhead, but it's simple enough to add some extra thickness to move the camera away if needed.

[image]

---

@Neil-YL I went ahead and ordered https://canada.newark.com/cyntech/lisiparoiwht-01/lisiparoi-white-led-camera-light/dp/31AC3472. Should ship relatively soon.

sgbaird, 2025-02-21:
Did the part come? If so, could you include a picture?

Neil-YL, 2025-02-21:
> Did the part come? If so, could you include a picture?

I think Jonathan had it in hand (weeks ago). 

I’ll look it up today. We’ve been asked to organize the stuffs in the training lab cubicles, and most things are currently stored in boxes.

sgbaird, 2025-02-25:
Could we try to get this attached to the camera head to see if there is clearance and visibility? The programming and power can be addressed later.

Neil-YL, 2025-02-26:
It seems the square hole on the light unit does not match the geometry of the camera len.

https://github.com/user-attachments/assets/3c5375a1-0b6d-4834-89d5-b26181808079 

The is the photo for the complete assembly from the guide: https://www.lisiparoi.co.uk/img/htu/complete.jpg

and it also cannot fit in the mount we have now.

Jonathan-Woo, 2025-02-26:
Yes this is something I noticed when trying to setup the light. I believe it's for an older version of the picam.

sgbaird, 2025-02-26:
To clarify, even if the mount didn't get in the way, the square hole is too small to fit over the silver box on the camera?

Neil-YL, 2025-02-26:
> To clarify, even if the mount didn't get in the way, the square hole is too small to fit over the silver box on the camera?

Yes, you can see from the video above that the square hole is too small to fit the lens, unlike in their final assembly photo.

sgbaird, 2025-02-26:
Yeah, not seeing great options.. https://www.google.com/search?q=raspberry+pi+camera+module+3+light+ring

Does the A1 mini have a built-in light? Will this be useful? Technically I think we could put it in the room with the OT-2 and leave the room lights on all the time (with some tape or something to say to leave it on). Might be an OK fallback..

Neil-YL, 2025-02-26:
> Yeah, not seeing great options.. https://www.google.com/search?q=raspberry+pi+camera+module+3+light+ring
>
> Does the A1 mini have a built-in light? Will this be useful? Technically I think we could [...]

I’m not entirely sure if I’m up to date with the latest plan, but I think the current plan is Jae’s students will only use the A1 Mini during working hours, so that either Sissi or I (or the course TA) can monitor the prints for spaghetti or adhesion failures. Hence, I think the night light set up is not very urgent (at least for the MO practical).

sgbaird, 2025-02-26:
Ok, we can come back to the lighting later.

> I think the current plan is Jae’s students will only use the A1 Mini during working hours, so that either Sissi or I (or the course TA) can monitor the prints for spaghetti or adhesion failures

I think spaghetti and adhesion failures are totally fine and necessary - the real issue is if these failures affect later iterations, hence all the discussion that I moved over to https://github.com/AccelerationConsortium/ac-training-lab/issues/156#issuecomment-2683252975. I think the students should be allowed to run after hours, but know that it might not be until the next day / after the weekend that something gets addressed if it goes wrong (i.e., "use after-hours at your own risk").

sgbaird, 2025-02-28:
Thanks for the context. Yes, I am aligned with Jae on the collection of data prior to student access, which is my request in #156 which is currently blocked by #160. I'll follow up in those issues.

SissiFeng, 2025-02-28:
I think I didn’t express it clearly. It’s not about just exploring possible parameter combinations in #156 , but providing a complete first-round training dataset.

Based on the new requirement this Wednesday:

Updated Experimental Requirements for MO:
  
1. Fixed Dataset for Initial Training
  Each student will be provided with a predefined dataset.  Students will train their models based on this dataset and propose the next batch of parameters to be tested.  

2. Model Evaluation and Selection  
  After students submit their models, all models will be evaluated.  The best-performing model will be selected. The selected model's proposed parameters will be used for the next round of testing (as the new constrains).  

3. Iterative Model Refinement
 The newly generated dataset (from the best model’s proposed parameters) will be fed back to the students.  Students will update and refine their models using the new data.  This process will continue iteratively, allowing models to compete and evolve.

SissiFeng, 2025-03-04:
When we tested the print script, we found after several collisions during the nozzle's part movement to the station for filament cutting, the camera mounted on the nozzle assembly is likely to shift from its original position. Over multiple impacts, the camera tends to tilt rather than remain perfectly vertical. This issue suggests that the mounting mechanism needs to be redesigned to enhance stability and prevent misalignment. I will try to video when this situation happen (not tilt every time, a little hard to capture).

Neil-YL, 2025-03-05:
> When we tested the print script, we found after several collisions during the nozzle's part movement to the station for filament cutting, the camera mounted on the nozzle assembly is likely to shift [...]

I have an ideal of adding some positioning/stopper on the current toolhead adaptor. I will print one to test.

sgbaird, 2025-03-10:
Great!

sgbaird, 2025-03-17:
At some point, let's modify the camera mount file to allow just a bit more room to remove the SD card without needing to disassemble the camera mount. When it was mounted upside down, it couldn't be removed. It looks like @jonathan-woo installed it in the orientation shown in the photo below to address this.

https://github.com/AccelerationConsortium/ac-training-lab/issues/11#issuecomment-2729500449

[image]

Neil-YL, 2025-03-17:
> At some point, let's modify the camera mount file to allow just a bit more room to remove the SD card without needing to disassemble the camera mount. When it was mounted upside down, it couldn't be removed. It looks like [@Jonathan-Woo](https://github.com/Jonathan-Woo) installed it in the orientation shown in the photo below to address this.
> 
> [#11 (comment)](https://github.com/AccelerationConsortium/ac-training-lab/issues/11#issuecomment-2729500449)
> 
> [image]
> 
> [@Neil-YL](https://github.com/Neil-YL) can you also enable and add the share link to the latest fusion 360 file for this? (Set as public access)

Mount: https://a360.co/41RtaGt 
Front cover: https://a360.co/4jq5PUd
The orientation of this Stream Pi is different from the OT-2 one because the FFC cable is not as long.

sgbaird, 2025-03-24:
Could benefit from a camera light:

(too close, and the toolhead casts a shadow)

I think offsets are in some kind of percent, where 0 is touching the bed.

offset = 10

[image]

offset = 25 (I think)
[image]

offset = 50 (not casting as large of a shadow)

[image]

sgbaird, 2025-03-25:
> > Did the part come? If so, could you include a picture?
> 
> I think Jonathan had it in hand (weeks ago).
> 
> I’ll look it up today. We’ve been asked to organize the stuffs in the training lab cubicles, and most things are currently stored in boxes.
> 
> ---update---
> 
> <img alt="Image" width="500" src="https://private-user-images.githubusercontent.com/179746567/415714363-d14ffbca-431f-4294-9a9d-901555734eac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI4NzMwNDgsIm5iZiI6MTc0Mjg3Mjc0OCwicGF0aCI6Ii8xNzk3NDY1NjcvNDE1NzE0MzYzLWQxNGZmYmNhLTQzMWYtNDI5NC05YTlkLTkwMTU1NTczNGVhYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyNVQwMzE5MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNDIwMmE2ZjgyZjAyZjVmYTBjZmNhZjY2MjkyYmEwNTEzZjMxZGQ3MmI4OWQ5OGJiNzEwMzlmYjNmMjkzMGUxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.9kmeiOlotYTQLl4HLQrAET_qGbIZ8ks9IkPJbAvGDLI"> <img alt="Image" width="500" src="https://private-user-images.githubusercontent.com/179746567/415715022-f3f66f8b-b9e4-4d1a-812b-e4ae805ad953.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI4NzMwNDgsIm5iZiI6MTc0Mjg3Mjc0OCwicGF0aCI6Ii8xNzk3NDY1NjcvNDE1NzE1MDIyLWYzZjY2ZjhiLWI5ZTQtNGQxYS04MTJiLWU0YWU4MDVhZDk1My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyNVQwMzE5MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMWZjNjc4Y2ZkMTI5NTgwMGVmNzBmZWQzNTg3NGQwNDY4NjJkYmQ0NTEwMmRlMDZkZDFmNGFmZWE5NzAyM2UxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sFko4HTDsM7NdBK9T8e2WVzLxjdKW_abk35BbLJSVzM"> <img alt="Image" width="500" src="https://private-user-images.githubusercontent.com/179746567/415714482-b9c38855-572c-4c49-983d-ae4bb44ff5aa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI4NzMwNDgsIm5iZiI6MTc0Mjg3Mjc0OCwicGF0aCI6Ii8xNzk3NDY1NjcvNDE1NzE0NDgyLWI5YzM4ODU1LTU3MmMtNGM0OS05ODNkLWFlNGJiNDRmZjVhYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyNVQwMzE5MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NWFhMzY1YzRhYWY2MDAxMDlhODNjNDRmNGI5M2Y3ZTI1MjQ1M2VjYTA1ZDQzNDFhNDQyNWY1ZTkxNjk2ZDRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.UUr-mln3lJawDyhMWFwTOS-ccgr82tOUrnAF-gpIR0k"> <img alt="Image" width="500" src="https://private-user-images.githubusercontent.com/179746567/415714593-184b695c-dce7-49c8-ae06-3d9f647daafa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI4NzMwNDgsIm5iZiI6MTc0Mjg3Mjc0OCwicGF0aCI6Ii8xNzk3NDY1NjcvNDE1NzE0NTkzLTE4NGI2OTVjLWRjZTctNDljOC1hZTA2LTNkOWY2NDdkYWFmYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyNVQwMzE5MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NWQ0MzUxZmIwOTllNDkwMmVjMzZiZjRhZDA3YTJlMzgwMDJiYTJkYzAzMTk4ZGM2MmVjNmM0NjM0OWM0OTMwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.QYXetv2BuWaeI0Pn4uwyWYQpNNI4KCMpJsWhy__9Q5o">
> Guide: https://www.lisiparoi.co.uk/how-to-use.htm

> It seems the square hole on the light unit does not match the geometry of the camera lens.
> 
>  IMG_2274.mov 
> The is the photo for the complete assembly from the guide: https://www.lisiparoi.co.uk/img/htu/complete.jpg
> 
> and it also cannot fit in the mount we have now.

Do the mounting holes line up for this? We could maybe deal with the size mismatch of the square. Could we also try hooking it up to see if we can control the lighting easily?

Neil-YL, 2025-03-25:
> > > Did the part come? If so, could you include a picture?
> > 
> > 
> > I think Jonathan had it in hand (weeks ago).
> > I’ll look it up today. We’ve been asked to organize the stuffs in the training lab cubicles, and most things are currently stored in boxes.
> > ---update---
> > <img alt="Image" width="500" src="https://private-user-images.githubusercontent.com/179746567/415714363-d14ffbca-431f-4294-9a9d-901555734eac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI4NzMwNDgsIm5iZiI6MTc0Mjg3Mjc0OCwicGF0aCI6Ii8xNzk3NDY1NjcvNDE1NzE0MzYzLWQxNGZmYmNhLTQzMWYtNDI5NC05YTlkLTkwMTU1NTczNGVhYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyNVQwMzE5MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNDIwMmE2ZjgyZjAyZjVmYTBjZmNhZjY2MjkyYmEwNTEzZjMxZGQ3MmI4OWQ5OGJiNzEwMzlmYjNmMjkzMGUxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.9kmeiOlotYTQLl4HLQrAET_qGbIZ8ks9IkPJbAvGDLI"> <img alt="Image" width="500" src="https://private-user-images.githubusercontent.com/179746567/415715022-f3f66f8b-b9e4-4d1a-812b-e4ae805ad953.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI4NzMwNDgsIm5iZiI6MTc0Mjg3Mjc0OCwicGF0aCI6Ii8xNzk3NDY1NjcvNDE1NzE1MDIyLWYzZjY2ZjhiLWI5ZTQtNGQxYS04MTJiLWU0YWU4MDVhZDk1My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyNVQwMzE5MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMWZjNjc4Y2ZkMTI5NTgwMGVmNzBmZWQzNTg3NGQwNDY4NjJkYmQ0NTEwMmRlMDZkZDFmNGFmZWE5NzAyM2UxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sFko4HTDsM7NdBK9T8e2WVzLxjdKW_abk35BbLJSVzM"> <img alt="Image" width="500" src="https://private-user-images.githubusercontent.com/179746567/415714482-b9c38855-572c-4c49-983d-ae4bb44ff5aa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI4NzMwNDgsIm5iZiI6MTc0Mjg3Mjc0OCwicGF0aCI6Ii8xNzk3NDY1NjcvNDE1NzE0NDgyLWI5YzM4ODU1LTU3MmMtNGM0OS05ODNkLWFlNGJiNDRmZjVhYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyNVQwMzE5MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NWFhMzY1YzRhYWY2MDAxMDlhODNjNDRmNGI5M2Y3ZTI1MjQ1M2VjYTA1ZDQzNDFhNDQyNWY1ZTkxNjk2ZDRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.UUr-mln3lJawDyhMWFwTOS-ccgr82tOUrnAF-gpIR0k"> <img alt="Image" width="500" src="https://private-user-images.githubusercontent.com/179746567/415714593-184b695c-dce7-49c8-ae06-3d9f647daafa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDI4NzMwNDgsIm5iZiI6MTc0Mjg3Mjc0OCwicGF0aCI6Ii8xNzk3NDY1NjcvNDE1NzE0NTkzLTE4NGI2OTVjLWRjZTctNDljOC1hZTA2LTNkOWY2NDdkYWFmYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMyNVQwMzE5MDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NWQ0MzUxZmIwOTllNDkwMmVjMzZiZjRhZDA3YTJlMzgwMDJiYTJkYzAzMTk4ZGM2MmVjNmM0NjM0OWM0OTMwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.QYXetv2BuWaeI0Pn4uwyWYQpNNI4KCMpJsWhy__9Q5o">
> > Guide: https://www.lisiparoi.co.uk/how-to-use.htm
> 
> > It seems the square hole on the light unit does not match the geometry of the camera lens.
> > IMG_2274.mov
> > The is the photo for the complete assembly from the guide: https://www.lisiparoi.co.uk/img/htu/complete.jpg
> > and it also cannot fit in the mount we have now.
> 
> Do the mounting holes line up for this? We could maybe deal with the size mismatch of the square, either ignoring the mismatch and mounting it anyway or cutting out a larger square. [@Neil-YL](https://github.com/Neil-YL) could you try mounting this tomorrow? Also, could you hooking it up to see if we can control the lighting easily?

> Wiring up the LISIPAROI
> The LISIPAROI has been designed to be powered from the Raspberry Pi providing enough power is available, if not a seperate power supply of 5 volts can be used.
> 
> Wiring up to the Raspberry Pi
> If you are using a power supply rated 2A or more, you can power your LISIPAROI directly from the Raspberry Pi.

I will see if can be power up by directly by the RPi zero2w

sgbaird, 2025-03-25:
We'll also probably be running it at relatively low power, and also turning it off when not in use.

Neil-YL, 2025-03-25:
Update:

Connect the 5v on the LISIPAROI to the 5v (pin 2) on the Zero 2 W

Connect GND (only 1 is needed) to the GND (pin 9) on the Zero 2 W

Connect GPIO to the 3v3 (pin 1) on the Zero 2 W

[image]

[image]

It needs the SD card and the system loaded in order to power up the lighting unit.  

[image]

Neil-YL, 2025-03-25:
>We'll also probably be running it at relatively low power, and also turning it off when not in use.

Note copied from the how to use page:

## Controlling from the Raspberry Pi
### Hooking up to a GPIO Pin
In this example GPIO 4 will be used to turn on the LISIPAROI, although any digital GPIO pin will work. Swap the 3v3 jumper wire to a chosen GPIO (or to 4 as pictured)

Switching the LISIPAROI on and off
Login remotely via PuTTY or start LXTerminal on the Raspberry Pi.

Create a new file by typing sudo nano flash.py  [Enter]

And paste the following code;

```
import RPi.GPIO as GPIO

import time
import picamera

GPIO.setmode(GPIO.BOARD)
GPIO.setup(4, GPIO.OUT)

GPIO.output(4, GPIO.HIGH)
with picamera.PiCamera() as camera:

camera.start_preview()
time.sleep(5)
camera.capture(‘/home/pi/Desktop/test.jpg’)
camera.stop_preview()

GPIO.output(4, GPIO.LOW)
```

 

It should look like this, just press CTRL + X to exit, press Y to save, and then Enter to confirm the filename.

code-start

 

To run the program, type in sudo python flash.py followed by enter. The LISIPAROI should switch on, a picture is taken. After 5 seconds the LISIPAROI will then switch off.

Neil-YL, 2025-03-25:
The example is for other version of RBPi. In our case I believe it will be GPIO 1?

[image]

sgbaird, 2025-03-25:
Oh, gotcha. Yeah, whatever works for powering it and still being able to do PWM. Just lmk what pin it's connected to once it's all set up

sgbaird, 2025-03-25:
E.g., GPIO18 (GPIO1 says no PWM)

[image]

https://pinouts.vercel.app/board/raspberry-pi-zero-2-w

Neil-YL, 2025-03-25:
I printed another new mount (may need refine design on the length of short side, seems to large compared to the previous design (to give room for the bolt from the cable)

[image]

Two M2 - 16 screws is use to assemble the Cam + mount + light unit.

[image]

GPIO connected to GPIO 18 (board pin 12) on the Zero2W

Some videos to test the on/off control and lighting level with PWM 

https://github.com/user-attachments/assets/738e6173-5e6c-44b7-bf86-45c4ffb30a8e
https://github.com/user-attachments/assets/37f3ad8a-ba58-4933-8255-524ec0f539f5

sgbaird, 2025-03-25:
Thank you! Tomorrow, could you move the light to the current A1 mini camera toolhead? Then we can test the light settings and distance offset in #156

sgbaird, 2025-03-27:
I swapped out the SD cards. Worked just fine, though now I'm thinking I should update the device serial in MongoDB 😅

sgbaird, 2025-04-04:
> I swapped out the SD cards. Worked just fine, though now I'm thinking I should update the device serial in MongoDB 😅

NOTE: [serial number is hard-coded to device](https://www.google.com/search?q=does+serial+change+if+you+change+OS+raspberry+pi), but hostname is specific to the SD card, so to keep things straight one has to manually keep track of which SD card corresponds to which serial.

sgbaird, 2025-06-13:
Including CAD file here (points to latest version): https://a360.co/4kbOj6e

[image]

For context, @xiaomguo is planning to help create a 3D printed mount to attach to a UR3e, basing it off this design (at least in terms of dimensions).