AccelerationConsortium/ac-dev-lab #159: Image transfer for A1-mini tool head camera

sgbaird, 2025-01-21:
MQTT might work for the image transfer. See for example https://www.reddit.com/r/esp32/comments/r6y28l/esp32_cam_how_to_send_good_quality_images_via_mqtt/, https://www.home-assistant.io/integrations/camera.mqtt/, etc. I think @kenzo-aspuru-takata dealt with sending images via MQTT with the OpenFlexure microscopes. This wouldn't really work for image streaming, but for point and shoot captures it's probably ok. However, while 256 MB/message upper limit probably won't be an issue, HiveMQ free tier is capped at 10 GB/month. Probably a major limiting factor for these workflows. https://www.google.com/search?q=hicemq+payload+limits. I've also recently cancelled the starter plan I had: #155 

I propose using Amazon S3 as outlined in the "tip" from https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.5-data-logging.html#additional-resources. i.e., send a command to capture the image via MQTT to the device, use the AWS API via the Zero 2W to upload to an S3 bucket (I need to get this set up on one of our accounts), and send the S3 URI back to the orchestrator via MQTT (possibly after double checking that it can be accessed, i.e., if there's some latency).

@Jonathan-Woo would you feel comfortable / have time to try this out within the next two days after getting the point and shoot working?

xref: https://ac-microcourses.readthedocs.io/en/latest/courses/hello-world/1.4.1-onboard-led-temp.html and corresponding assignment, and see tip from 1.5 mentioned above for AWS links

sgbaird, 2025-01-21:
xref: #144

sgbaird, 2025-01-23:
@Jonathan-Woo any updates from what you worked on yesterday? (Again, np if it goes into next week)

Jonathan-Woo, 2025-01-24:
Finished. PR here https://github.com/AccelerationConsortium/ac-training-lab/pull/171

Jonathan-Woo, 2025-02-07:
Forgot to include a demo:

https://github.com/user-attachments/assets/99868302-46eb-47b6-a491-6f40404063e2

Jonathan-Woo, 2025-02-07:
Might need to tune the image file size as it's currently capturing uncompressed PNGs which are 35 MB each.

sgbaird, 2025-02-21:
Good point, 35 MB is a lot. For the intended use case, we probably only need a few MB worth. Feel free to add some compression.

I also noticed your comment about S3 object lifetimes. Ideally, we'd want to avoid deleting the images at least until we can upload a large batch of them to Zenodo or similar as a fixed snapshot.

sgbaird, 2025-02-25:
It seems the MWE is complete and tested (thank you!). Is the camera currently set up to receive requests for the image capture?

Jonathan-Woo, 2025-02-28:
xref https://github.com/AccelerationConsortium/ac-training-lab/issues/147#issuecomment-2691703083

Jonathan-Woo, 2025-03-06:
After running for a few days I'm getting this error:

```
[119:01:07.780090161] [10084]  WARN V4L2 v4l2_videodevice.cpp:2095 /dev/video0[17:cap]: Dequeue timer of 1000000.00us has expired!
[119:01:07.811791412] [10084] ERROR RPI pipeline_base.cpp:1364 Camera frontend has timed out!
[119:01:07.811912193] [10084] ERROR RPI pipeline_base.cpp:1365 Please check that your camera sensor connector is attached securely.
[119:01:07.811981829] [10084] ERROR RPI pipeline_base.cpp:1366 Alternatively, try another cable and/or sensor.
```

Only fixed through rebooting the rpi. Maybe we should have scheduled downtimes where the A1 camera and the live streams can reboot daily.

sgbaird, 2025-03-10:
Could you look through the following, see if you find anything?

- https://github.com/raspberrypi/picamera2/issues/854
- https://forums.raspberrypi.com/viewtopic.php?t=336785
- https://github.com/raspberrypi/picamera2/issues/952

A scheduled reboot sounds good (maybe 2 am ET or something).

To clarify, was this for the "point and shoot" toolhead camera or the livestream?

Jonathan-Woo, 2025-03-10:
Yes this was for the point and shoot toolhead camera.

sgbaird, 2025-03-10:
Ah, got it. Lmk if you find anything useful in the links from above. I'm also OK with setting up a daily reboot. Tbh I'm not sure we need to even do anything on the HF or Colab side (at least for now). They will just receive a timeout error I think.

sgbaird, 2025-03-10:
Potentially some discussion already related to implementing RPi reboots: https://github.com/AccelerationConsortium/ac-training-lab/issues?q=is%3Aissue%20state%3Aopen%20reboot

sgbaird, 2025-03-10:
Is the script set to auto-start (i.e., if I unplug it and plug it back in, will it go back up?)

Jonathan-Woo, 2025-03-10:
No sorry.

ssh ***@***.***
Password: password

cd to Desktop/actraininglab/src/actraininglab/bambu_cam/

Activate the venv and run device.py

Sorry I’ll clean it up!

sgbaird, 2025-03-10:
Thanks! Planning to go in tomorrow to implement the integration

sgbaird, 2025-03-14:
Potential S3 bucket settings:

[image]

Though it seems like maybe I need to uncheck all public access restrictions (can't update the policy per recommendation in [transcript](https://chatgpt.com/share/fb6746ca-3791-49a3-b83c-9f91c350796b)):

[image]

Recommendation by transcript for policy:

```
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Sid": "PublicRead",
            "Effect": "Allow",
            "Principal": "*",
            "Action": "s3:GetObject",
            "Resource": "arn:aws:s3:::rpi-zero2w-toolhead-camera/*"
        }
    ]
}
```

@Jonathan-Woo could you comment on anything I'm missing here or that you did differently?

sgbaird, 2025-03-14:
@Jonathan-Woo mentioned that the access key is added to ~/.aws/credentials on the A1 cam. I might adjust the code to pass it in explicitly as env var parameters.

@Jonathan-Woo what policy did you use for this user? (assuming it's attached to a new user and not to root)

Jonathan-Woo, 2025-03-14:
```
{
    "Version": "2012-10-17",
    "Statement": [
        {
            "Sid": "FullAccess",
            "Effect": "Allow",
            "Principal": {
                "AWS": "arn:aws:iam::160885275440:user/jwoo"
            },
            "Action": "s3:*",
            "Resource": "arn:aws:s3:::jwoo-picam-bucket"
        }
    ]
}
```

sgbaird, 2025-03-14:
Let's go with the one you shared

sgbaird, 2025-03-17:
We switched back to the former one, since the latter one seems to allow deletion of the bucket, among other actions.

sgbaird, 2025-03-17:
Note to self: add sortable timestamp int and human-readable timestamp to payload: 

Per example in: https://github.com/AccelerationConsortium/ac-training-lab/blob/75e44ec381746a3c58db2d132de4cb8e189e7b27/src/ac_training_lab/A1-cam/device.py#L50-L55

https://github.com/sparks-baird/self-driving-lab-demo/blob/c69953a55ede02d4d95ff9660b8093dca0e155db/src/public_mqtt_sdl_demo/lib/data_logging.py#L143-L151

closed 2025-03-19 by sgbaird