AccelerationConsortium/ac-dev-lab #161: Equipment monitoring camera for A1 mini (standalone, separate from toolhead camera)

sgbaird, 2025-01-21:
The built-in camera on the A1 mini doesn't support streaming directly (see below), so in addition to the "toolhead camera" from #158, we'll likely want to have a separate monitoring camera that lets people see the device in motion to create the "feeling of control remote hardware". This will be the same as any other setup described in #11. Might seem a bit redundant given there's already a toolhead camera and the built-in A1 mini camera, but I think it will enhance the experience to be able to observe the full system operating in response to commands, and possibly for troubleshooting too.

---

### Some resources related to the Bambu Lab built-in camera / A1 Mini streaming being unsupported:
- https://wiki.bambulab.com/en/software/bambu-studio/virtual-camera
- https://github.com/bambulab/BambuStudio/issues/1536#issuecomment-2469180080
(someone got A1 mini working, but it's not stable / requires restarts)
- https://github.com/synman/webcamd | https://github.com/synman/bambu-go2rtc
both found via [reddit](https://www.reddit.com/r/BambuLab/comments/191nqdq/a1_webcam_streaming/)

Maybe webcamd from above would work, but seems like it could be more work than its worth compared with replicating the Pi Zero 2W setup.

---

@Jonathan-Woo could you also get one of these cameras set up, mounted, and pointing at the A1 mini? (secondary priority to the image capture task mentioned in #144)

sgbaird, 2025-02-21:
Thanks for all the effort getting the OT-2 livestream set up (and especially all the ground work leading up to it). What's the status on setting up this one?

Jonathan-Woo, 2025-02-22:
I was busy with the toolhead camera but this should be doable. The only concern is the upload bandwidth if we have to use the Nokia. @Neil-YL any luck with using utoronto wifi?

Neil-YL, 2025-02-24:
> I was busy with the toolhead camera but this should be doable. The only concern is the upload bandwidth if we have to use the Nokia. [@Neil-YL](https://github.com/Neil-YL) any luck with using utoronto wifi?

Not yet. Last week the students from Jae's course were using the system actively. I will have a try tomorrow.

sgbaird, 2025-02-24:
I think it should be straightforward using network manager. Could you remind me - are we running this in headless mode or with a GUI-based OS? If GUI-based, I think the WiFi icon at the top-right will let you do it. If in headless, see https://openflexure.discourse.group/t/connecting-to-wpa2-enterprise-networks-e-g-eduroam-university-settings/1659/2 and https://www.google.com/search?q=connect+to+eduroam+on+raspberry+pi+network+manager

Neil-YL, 2025-02-24:
> I think it should be straightforward using network manager. Could you remind me - are we running this in headless mode or with a GUI-based OS? If GUI-based, I think the WiFi icon at the top-right [...]

Do you think it is possible to SSH to the device once it is connected to UofT/Eduroam network?

sgbaird, 2025-02-24:
Oh, good point. I'm not sure. In the meantime, if we have trouble with eduroam / UoT, could we reduce the resolution to the lowest reasonable/allowed setting for YouTube, which we could increase later on? (I.e., just to get more pieces connected). Thanks so much for working on all these pieces ☺️

Neil-YL, 2025-02-25:
> Oh, good point. I'm not sure. In the meantime, if we have trouble with eduroam / UoT, could we reduce the resolution to the lowest reasonable/allowed setting for YouTube, which we could increase later on? (I.e., just to get more pieces connected). Thanks so much for working on all these pieces ☺️

My plan is to set the streaming python file as auto-started and leave it with UofT network to test the stability. I will test it today.

sgbaird, 2025-02-25:
That's great news! We can try to get a non-personal account set up with it. I need to explore that - there may be someone else at the AC who's done that. Is the one you're talking about the OT-2 stream or the A1 mini stream? If the latter, could you add a link to the stream?

Neil-YL, 2025-02-25:
I just updated the stream for the OT-2. Do we have another mount for the Picam? (one is used for the OT-2 streaming and one is used for the toolhead cam of the Bambu)

BTW today we have a meeting with Jae and I believe we will move forwards with the Bambu A1 mini for the MO practical.

sgbaird, 2025-02-25:
Looks like we have 3, though I'm not sure where the 3rd would be.

[image]

Thanks for the update about the meeting! I think the office (paper) printer is connected over Ethernet. The office building has a lot of restrictions about enabling Ethernet ports. What I'm pretty sure about is that the AC can get additional UToronto emails that aren't tied to an individual. Just not sure if those emails can be used to sign into the network..

Neil-YL, 2025-02-26:
> Looks like we have 3, though I'm not sure where the 3rd would be.
>
> [image]
>
> Thanks for the update about the meeting! I think the office (paper) printer is connected over Ethernet. The office [...]

I will look it up

sgbaird, 2025-02-26:
Ah, got it. If you gave one of the mounts to him, then I don't think there are more. Maybe at NRC, I'd need to check with him once he's back from travel.

If you could get the livestream set up, even without the mount yet, that would be great to keep getting more pieces connected. You can cover it with a box or something if needed since it won't be mounted yet.

You could also give it a try with the camera case shown above. Pretty sure that hasn't been used yet, though I don't think it was designed to be attached to a tripod.

Printing something simple sounds find. The main issue is with the nuts and bolts. I'm not sure what we have on hand. Ideally, we'd use the same nuts and bolts as what's from the "Pro Mini Camera Mount".

Neil-YL, 2025-02-26:
> Ah, got it. If you gave one of the mounts to him, then I don't think there are more. Maybe at NRC, I'd need to check with him once he's back from travel.
>
> If you could get the livestream set up, [...]

I got the Pizero2 and Picam ready. Have some troubles on try to using SD card cloned from the OT-2 streaming PiZero2. I will figure it out tomorrow and start a new livestream on AC hardware YouTube.

sgbaird, 2025-02-26:
I think typically setting up the Pi Zero 2W from scratch, and then following setup instructions for the streaming / downloading necessary files / installing necessary packages / etc. is best. If we don't have documentation that makes it easy to replicate, then we should add some. @Jonathan-Woo are there instructions somewhere? I know I need to update the website to have a list of devices to make things easier to find and to link the device pages to.

(even if we can clone the SD card because we happen to have it physically, that's not very replicable for others)

Jonathan-Woo, 2025-02-26:
Not yet, I can add a README to the source code.

Neil-YL, 2025-02-27:
[image]

The Bambu A1 Mini stream is live now (currently showing a grey screen of my desk). I’ll print a Picam mount and set it up in place.

Sissi said she would put the embeded live stream on HF as well.

SissiFeng, 2025-02-27:
[image]

The bambu space is updated (embedding the youtube stream address).

Neil-YL, 2025-02-27:
Job done.

[image]

[image]

sgbaird, 2025-02-28:
Any idea what happened with the livestream? I saw that there is a new one up (thank you).

I'm interested to know if there were any logs or indications of why it stopped (e.g., network connectivity, Pi going unresponsive, etc.)

Neil-YL, 2025-03-03:
My speculation is that the issue is caused by either cabling blocking the movement or the sensor on the Bambu. I will troubleshoot it with Sissi.

Neil-YL, 2025-03-03:
disassembled the cutter and reassembled it, also checking that the gears work fine as suggested in the manual. Now, we can print some files storaged on the machine, but our customized G-code still fails, so we’re looking into that.

sgbaird, 2025-03-03:
If you take the camera off (i.e., swap out with the default cover), does the g code work ok?

SissiFeng, 2025-03-03:
This issue was an unexpected incident this morning. After our troubleshooting, we have restored the progress from last Friday. The root cause remains unknown.

sgbaird, 2025-03-03:
Thanks for the update!

sgbaird, 2025-03-05:
Updated the stream link: https://huggingface.co/spaces/AccelerationConsortium/bambu-a1-mini. Thanks for resetting the stream when it went down

SissiFeng, 2025-03-05:
The stream dropped out when I checked in the morning. The reason may be that HF Spaces is a stateless web application and does not support websocket long connections. Many operations, such as page refreshes, will cause HF to trigger the idle mechanism, which kills processes.

sgbaird, 2025-03-11:
I unwittingly unplugged it and plugged it back in, but looks like the stream went down already.

sgbaird, 2025-03-15:
We killed the stream again.. @Neil-YL could you set back up when you get a chance?

Neil-YL, 2025-03-17:
> We killed the stream again.. [@Neil-YL](https://github.com/Neil-YL) could you set back up when you get a chance?

Has the stream pi been reset?

Neil-YL, 2025-03-17:
Not sure why I cannot start the streaming, here is the log:
```
[0:10:23.652294255] [803] ^[[1;32m INFO ^[[1;37mCamera ^[[1;34mcamera_manager.cpp:327 ^[[0mlibcamera v0.4.0+53-29156679
[0:10:24.031715513] [814] ^[[1;33m WARN ^[[1;37mRPiSdn ^[[1;34msdn.cpp:40 ^[[0mUsing legacy SDN tuning - please consider moving SDN inside rpi.deno>
[0:10:24.037262019] [814] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mvc4.cpp:447 ^[[0mRegistered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device />
[0:10:24.037424305] [814] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mpipeline_base.cpp:1121 ^[[0mUsing configuration file '/usr/share/libcamera/pipeline/rpi>
[0:10:24.060916060] [803] ^[[1;32m INFO ^[[1;37mCamera ^[[1;34mcamera.cpp:1202 ^[[0mconfiguring streams: (0) 1280x720-XBGR8888 (1) 1536x864-SRGGB10>
[0:10:24.061700926] [814] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mvc4.cpp:622 ^[[0mSensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 15>
default: No such process
```

@Jonathan-Woo Does it require any configuration of the Picam? What I have installed: 
picamera2          0.3.25
wget               3.2
ffmpeg version 5.1.6-0+deb12u1+rpt1 Copyright (c) 2000-2024 the FFmpeg developers
built with gcc 12 (Debian 12.2.0-14)

Jonathan-Woo, 2025-03-17:
> Not sure why I cannot start the stream, here is the log:

Have you tried rebooting the pi? Picam is sometimes fickle.

[0:10:24.037262019] [814] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mvc4.cpp:447 ^[[0mRegistered camera ***@***.******@***.*** to Unicam device />
[0:10:24.061700926] [814] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mvc4.cpp:622 ^[[0mSensor: ***@***.******@***.*** - Selected sensor format: 15>

Not sure why I cannot start the stream, here is the log:

[0:10:24.037262019] [814] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mvc4.cpp:447 ^[[0mRegistered camera ***@***.******@***.*** to Unicam device />
[0:10:24.061700926] [814] ^[[1;32m INFO ^[[1;37mRPI ^[[1;34mvc4.cpp:622 ^[[0mSensor: ***@***.******@***.*** - Selected sensor format: 15>

Neil-YL, 2025-03-17:
> Have you tried rebooting the pi? Picam is sometimes fickle.
> […](#)

Reboot several times. Same "No such process" log above
(sorry for accidentally editing your reply)

sgbaird, 2025-03-19:
Something to do with the switch to RPi OS Lite? (I think you made this switch already @Jonathan-Woo). Thoughts?

Jonathan-Woo, 2025-03-19:
Have you tried recreating the venv?

Neil-YL, 2025-03-19:
> Is this running on OS Lite now or still on the Desktop version?

I think it is on command line interface now

sgbaird, 2025-03-20:
https://github.com/AccelerationConsortium/ac-training-lab/pull/199

See also [README](https://github.com/AccelerationConsortium/ac-training-lab/blob/main/src/ac_training_lab/picam/README.md) and latest [device.py](https://github.com/AccelerationConsortium/ac-training-lab/blob/main/src/ac_training_lab/picam/device.py) (renamed from stream.py)

At the moment, broadcasting to: https://www.youtube.com/live/ix5C96OImlM

Neil-YL, 2025-03-20:
> See also [README](https://github.com/AccelerationConsortium/ac-training-lab/blob/main/src/ac_training_lab/picam/README.md) and latest [device.py](https://github.com/AccelerationConsortium/ac-training-lab/blob/main/src/ac_training_lab/picam/device.py) (renamed from stream.py)
> 
> At the moment, broadcasting to: https://www.youtube.com/live/ix5C96OImlM
> 
> Context and decisions in part documented at [#199](https://github.com/AccelerationConsortium/ac-training-lab/pull/199)

Should I mimic the set up for the OT-2 streaming pi?

sgbaird, 2025-03-20:
Yes, that would be great. It would be a good test of the steps I wrote, too.

Neil-YL, 2025-03-21:
> Yes, that would be great. It would be a good test of the steps I wrote, too.

Did you get the non-personal UofT network account work? I believe I need to reflash the pizero to the headless OS and set up new network login

sgbaird, 2025-03-21:
Not yet, I'll ask about it today

sgbaird, 2025-03-21:
Looks like it will be difficult to have one that's not tied to a personal account. Maybe something with eduroam could work out.

https://security.utoronto.ca/services/utorid/usage-guide/

I looked into starlink too but I think it's 5 MB/s upload speed, which isn't much better than the Nokia.

Neil-YL, 2025-03-21:
> Yes, that would be great. It would be a good test of the steps I wrote, too.

I went through this [readme file](https://github.com/AccelerationConsortium/ac-training-lab/blob/main/src/ac_training_lab/picam/README.md).

I believe ffmpeg should be added in the dependencies: 

> FileNotFoundError: [Errno 2] No such file or directory: 'ffmpeg'

`
sudo apt install ffmpeg --no-install-recommends`

I started the stream with the system service and seems all good:

[image]

I will mount it back to and to see if the autostart works

sgbaird, 2025-04-01:
Ah, yes! If you could update please that would be great. I will likely try to avoid duplication in the docs somehow more automatically.

sgbaird, 2025-04-02:
Could you check the logs and paste here?

Neil-YL, 2025-04-02:
> Could you check the logs and paste here?

I think the A1 mini stream on Youtube went down earlier than the 2am restart so the most recent log is the 2am restart and no errors were caught up. 

but the one for the OT2 went down as well and luckily it is after 2am so here are the logs:

> Apr 02 03:37:16 rpi-zero2w-ot2-stream python3[602]: [48.0K blob data]
> Apr 02 03:41:24 rpi-zero2w-ot2-stream python3[602]: [48.0K blob data]
> Apr 02 03:45:32 rpi-zero2w-ot2-stream python3[602]: [48.0K blob data]
> Apr 02 03:49:40 rpi-zero2w-ot2-stream python3[602]: [48.0K blob data]
> Apr 02 03:53:48 rpi-zero2w-ot2-stream python3[602]: [48.0K blob data]
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]: [21.7K blob data]
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]:     Last message repeated 2 times
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]: [flv @ 0x559c603850] Failed to update header with correct duration.
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]: [flv @ 0x559c603850] Failed to update header with correct filesize.
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]: Error writing trailer of rtmp://a.rtmp.youtube.com/live2/ydj7-frx3-u6p3-8z4z-f6jp: Broken pipe
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]: frame=2064915 fps= 15 q=-1.0 Lsize=10415517kB time=38:14:08.56 bitrate= 619.9kbits/s speed=   1x
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]: video:10241658kB audio:35119kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 1.350034%
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]: Error closing file rtmp://a.rtmp.youtube.com/live2/ydj7-frx3-u6p3-8z4z-f6jp: Broken pipe
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]: [aac @ 0x559c5a1770] Qavg: 65536.000
> Apr 02 03:55:40 rpi-zero2w-ot2-stream python3[602]: Conversion failed!
> Apr 02 03:55:44 rpi-zero2w-ot2-stream systemd[1]: picam.service: Deactivated successfully.
> Apr 02 03:55:44 rpi-zero2w-ot2-stream systemd[1]: picam.service: Consumed 10h 4min 49.609s CPU time.

Neil-YL, 2025-04-03:
Both of the stream went down yesterday.
Logs on OT-2: 
> Apr 03 01:15:32 rpi-zero2w-ot2-stream python3[603]: [48.0K blob data]
> Apr 03 01:19:42 rpi-zero2w-ot2-stream python3[603]: [48.0K blob data]
> Apr 03 01:23:52 rpi-zero2w-ot2-stream python3[603]: [47.9K blob data]
> Apr 03 01:28:03 rpi-zero2w-ot2-stream python3[603]: [48.0K blob data]
> Apr 03 01:32:13 rpi-zero2w-ot2-stream python3[603]: [48.0K blob data]
> Apr 03 01:34:19 rpi-zero2w-ot2-stream python3[603]: [24.3K blob data]
> Apr 03 01:34:19 rpi-zero2w-ot2-stream python3[603]:     Last message repeated 2 times
> Apr 03 01:34:20 rpi-zero2w-ot2-stream python3[603]: [flv @ 0x559e2ea0a0] Failed to update header with correct duration.
> Apr 03 01:34:20 rpi-zero2w-ot2-stream python3[603]: [flv @ 0x559e2ea0a0] Failed to update header with correct filesize.
> Apr 03 01:34:20 rpi-zero2w-ot2-stream python3[603]: Error writing trailer of rtmp://a.rtmp.youtube.com/live2/ydj7-frx3-u6p3-8z4z-f6jp: Broken pipe
> Apr 03 01:34:20 rpi-zero2w-ot2-stream python3[603]: frame=600045 fps= 15 q=-1.0 Lsize= 3339119kB time=11:06:50.30 bitrate= 683.7kbits/s speed=   1x
> Apr 03 01:34:20 rpi-zero2w-ot2-stream python3[603]: video:3288592kB audio:10208kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 1.222236%
> Apr 03 01:34:20 rpi-zero2w-ot2-stream python3[603]: Error closing file rtmp://a.rtmp.youtube.com/live2/ydj7-frx3-u6p3-8z4z-f6jp: Broken pipe
> Apr 03 01:34:20 rpi-zero2w-ot2-stream python3[603]: [aac @ 0x559e289190] Qavg: 65536.000
> Apr 03 01:34:20 rpi-zero2w-ot2-stream python3[603]: Conversion failed!
> Apr 03 01:34:23 rpi-zero2w-ot2-stream systemd[1]: picam.service: Deactivated successfully.
> Apr 03 01:34:23 rpi-zero2w-ot2-stream systemd[1]: picam.service: Consumed 2h 56min 43.952s CPU time.

Here is the health check monitor log:
[image]

Logs on A1mini:

> 
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream python3[857]:     Last message repeated 2 times
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream python3[857]: [flv @ 0x557c41a020] Failed to update header with correct duration.
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream python3[857]: [flv @ 0x557c41a020] Failed to update header with correct filesize.
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream python3[857]: Error writing trailer of rtmp://a.rtmp.youtube.com/live2/902j-wzqz-sdz1-7de6-fpc0: Broken p>
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream python3[857]: size=      34kB time=00:00:35.22 bitrate=   8.0kbits/s speed=13.9x
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream python3[857]: video:0kB audio:9kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 282.>
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream python3[857]: Error closing file rtmp://a.rtmp.youtube.com/live2/902j-wzqz-sdz1-7de6-fpc0: Broken pipe
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream python3[857]: [aac @ 0x557c41a8a0] Qavg: 65536.000
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream python3[857]: Conversion failed!
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream systemd[1]: picam.service: Deactivated successfully.
> Apr 02 14:29:10 rpi-zero2w-a1mini-stream systemd[1]: picam.service: Consumed 4.405s CPU time.
> Apr 02 14:29:20 rpi-zero2w-a1mini-stream systemd[1]: picam.service: Scheduled restart job, restart counter is at 10.
> Apr 02 14:29:20 rpi-zero2w-a1mini-stream systemd[1]: Stopped picam.service - Start picam device.py script.
> Apr 02 14:29:20 rpi-zero2w-a1mini-stream systemd[1]: picam.service: Consumed 4.405s CPU time.
> Apr 02 14:29:20 rpi-zero2w-a1mini-stream systemd[1]: picam.service: Start request repeated too quickly.
> Apr 02 14:29:20 rpi-zero2w-a1mini-stream systemd[1]: picam.service: Failed with result 'start-limit-hit'.
> Apr 02 14:29:20 rpi-zero2w-a1mini-stream systemd[1]: Failed to start picam.service - Start picam device.py script.

No sure if I understand it properly but I feel like the service has not been restarted on OT2?

sgbaird, 2025-04-03:
Hmm.. `systemd[1]: Failed to start picam.service - Start picam device.py script.` on A1 mini but not OT-2.

Btw is this piece sensitive? `rtmp://a.rtmp.youtube.com/live2/902j-wzqz-sdz1-7de6-fpc0`? I didn't think about it until now, just noticed.

Neil-YL, 2025-04-03:
> Hmm.. `systemd[1]: Failed to start picam.service - Start picam device.py script.` on A1 mini but not OT-2.
> 
> Btw is this piece sensitive? `rtmp://a.rtmp.youtube.com/live2/902j-wzqz-sdz1-7de6-fpc0`? I didn't think about it until now, just noticed.

It is the stream key of the stream broadcast, we can reissue one to our device later if we have any concerns on leaked stream key

sgbaird, 2025-04-03:
Ok, just checking. At some point soon let's refresh the stream key and keep in mind for later (e.g., redacting when posting logs). I probably posted some with stream keys unwittingly already 😅

Since I think picamera is the one issuing the output, probably not worth trying to hide at the log level.

Neil-YL, 2025-04-03:
> Can you have a look at this PR? https://github.com/AccelerationConsortium/ac-training-lab/pull/217/files

Sure. I can apply the change on the devices and start the stream.

Neil-YL, 2025-04-03:
https://github.com/user-attachments/assets/7d50fea4-0d39-40ad-965f-1ac9a1d2dd4b

Lucky to capture a No data on YouTube but no restart FFmpeg or Picam.service.

From more detailed logs:

> [flv @ 0x558f882400] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly

I added [`"-fflags", "+genpts"`](https://video.stackexchange.com/questions/24462/timestamps-are-unset-in-a-packet-for-stream-0-this-is-deprecated-and-will-stop) under the first line of the ffmpeg setup, seems to remove this warning currently. 

> frame= 4458 fps= 15 q=-1.0 size=   36670kB time=00:05:07.26 bitrate= 976.1kbits/s speed=1.04x    
> frame= 8250 fps= 15 q=-1.0 size=   67855kB time=00:09:20.69 bitrate= 991.4kbits/s speed=1.02x    
> frame=12051 fps= 15 q=-1.0 size=   99119kB time=00:13:34.11 bitrate= 997.3kbits/s speed=1.02x    
> frame=15852 fps= 15 q=-1.0 size=  130321kB time=00:17:46.97 bitrate=1000.6kbits/s speed=1.01x    
> frame=19647 fps= 15 q=-1.0 size=  161576kB time=00:22:00.42 bitrate=1002.4kbits/s speed=1.01x    
> frame=21715 fps= 14 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.983x    
> frame=21715 fps= 12 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.847x    
> frame=21715 fps= 11 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.744x    
> frame=21715 fps=9.4 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.664x    
> frame=21715 fps=8.5 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.599x    
> frame=21715 fps=7.7 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.545x    
> frame=21715 fps=7.1 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.501x    
> frame=21715 fps=6.6 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.463x    
> frame=21715 fps=6.1 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.43x     
> frame=21715 fps=5.7 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.402x    
> frame=21715 fps=5.3 q=-1.0 size=  178668kB time=00:25:34.14 bitrate= 954.0kbits/s speed=0.377x

From logs above, the frame and the time keep the same for a long time, which indicates that ffmpeg keeps having the same frame and same timestamp, could be libcamera-vid not sending new frame or something wrong between two. However, neither of them will trigger the restart on the device.py so maybe this is why YouTube kills the stream eventually.

sgbaird, 2025-04-03:
Good point. Maybe we need a separate way to trigger a restart from the device.py side (and also makes sense why the while loop wouldn't help)

sgbaird, 2025-04-03:
I'll try to think of where we might be able to ask for some help on this (e.g., some kind of ffmpeg forum). Honestly might be worth posting a question on stack exchange.

sgbaird, 2025-04-04:
See updated comment above

Neil-YL, 2025-04-07:
Current device.py running on OT2 pizero2w (I believe it is still the old version of the stream code with the P1/P2 process for libcamera-vid /FFmpeg)seems promising in handling broken pipe error. At lease the live stream survived since last Thursday and there are very recent logs on restarting the script when broken pipe happened:

 
> frame=414165 fps= 15 q=-1.0 size= 1883958kB time=07:40:16.69 bitrate= 558.8kbits/s speed=   1x    
> frame=417923 fps= 15 q=-1.0 size= 1901917kB time=07:44:26.66 bitrate= 559.1kbits/s speed=   1x    
> av_interleaved_write_frame(): Connection reset by peer:47.68 bitrate= 559.2kbits/s speed=0.998x    
>     Last message repeated 1 times
> [flv @ 0x55686ebb60] Failed to update header with correct duration.
> [flv @ 0x55686ebb60] Failed to update header with correct filesize.
> Error writing trailer of rtmp://a.rtmp.youtube.com/live2/ydj7-frx3-u6p3-8z4z-f6jp: Connection reset by peer
> frame=419131 fps= 15 q=-1.0 Lsize= 1907611kB time=07:45:47.74 bitrate= 559.2kbits/s speed=0.998x
> video:1872318kB audio:7130kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 1.498453%
> Error closing file rtmp://a.rtmp.youtube.com/live2/ydj7-frx3-u6p3-8z4z-f6jp: Broken pipe
> [aac @ 0x5568683f50] Qavg: 65536.000
> Conversion failed!
> ffmpeg version 5.1.6-0+deb12u1+rpt1 Copyright (c) 2000-2024 the FFmpeg developers
>   built with gcc 12 (Debian 12.2.0-14)
>   configuration: --prefix=/usr --extra-version=0+deb12u1+rpt1 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --disable-mmal --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sand --enable-sdl2 --disable-sndio --enable-libjxl --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-librav1e --enable-shared
>   libavutil      57. 28.100 / 57. 28.100
>   libavcodec     59. 37.100 / 59. 37.100
>   libavformat    59. 27.100 / 59. 27.100
>   libavdevice    59.  7.100 / 59.  7.100
>   libavfilter     8. 44.100 /  8. 44.100
>   libswscale      6.  7.100 /  6.  7.100
>   libswresample   4.  7.100 /  4.  7.100
>   libpostproc    56.  6.100 / 56.  6.100
> Input #0, lavfi, from 'anullsrc=channel_layout=stereo:sample_rate=44100':
>   Duration: N/A, start: 0.000000, bitrate: 705 kb/s
>   Stream #0:0: Audio: pcm_u8, 44100 Hz, stereo, u8, 705 kb/s
> Input #1, h264, from 'pipe:0':
>   Duration: N/A, start: 1744033633.143562, bitrate: N/A
>   Stream #1:0: Video: h264 (High), yuv420p(progressive), 854x480, 15 fps, 15 tbr, 1200k tbn
> Stream mapping:
>   Stream #1:0 -> #0:0 (copy)
>   Stream #0:0 -> #0:1 (pcm_u8 (native) -> aac (native))
> Output #0, flv, to 'rtmp://a.rtmp.youtube.com/live2/ydj7-frx3-u6p3-8z4z-f6jp':
>   Metadata:
>     encoder         : Lavf59.27.100
>   Stream #0:0: Video: h264 (High) ([7][0][0][0] / 0x0007), yuv420p(progressive), 854x480, q=2-31, 15 fps, 15 tbr, 1k tbn
>   Stream #0:1: Audio: aac (LC) ([10][0][0][0] / 0x000A), 44100 Hz, stereo, fltp, 128 kb/s
>     Metadata:
>       encoder         : Lavc59.37.100 aac
> frame= 3821 fps= 15 q=-1.0 size=   17929kB time=00:04:12.70 bitrate= 581.5kbits/s speed=   1x    
> frame= 7619 fps= 15 q=-1.0 size=   35815kB time=00:08:25.38 bitrate= 580.5kbits/s speed=   1x    
> frame=11408 fps= 15 q=-1.0 size=   53781kB time=00:12:38.41 bitrate= 580.9kbits/s speed=   1x    
> frame=15207 fps= 15 q=-1.0 size=   71564kB time=00:16:51.08 bitrate= 579.8kbits/s speed=   1x    
> frame=18996 fps= 15 q=-1.0 size=   89420kB time=00:21:04.18 bitrate= 579.4kbits/s speed=   1x  

Not sure if it can handle the frozen frame tho.

sgbaird, 2025-04-07:
Let's set up this script from above https://github.com/AccelerationConsortium/ac-training-lab/issues/161#issuecomment-2776973256, on a new Pi Zero 2W

Neil-YL, 2025-04-07:
Have been working on the new script on Zero2w-Test, since it is on UofT and the mycobot-280 is still on Nokia. 

The latest version of the device.py:
```

import subprocess
import threading
import time
import logging
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput
from my_secrets import STREAM_KEY, STREAM_URL
# ====== CONFIGURATION ======
YOUTUBE_URL = STREAM_URL
YOUTUBE_KEY = STREAM_KEY
STREAM_URL = f"{YOUTUBE_URL}/{YOUTUBE_KEY}"

RESOLUTION = (1280, 720)
FRAMERATE = 15
BACKOFF_SECONDS = 5
FRAME_TIMEOUT = 10  # seconds before watchdog triggers

# ====== LOGGING SETUP ======
logging.basicConfig(
    level=logging.INFO,
    format="%(asctime)s [%(levelname)s] %(message)s",
    handlers=[
        logging.FileHandler("livestream.log"),
        logging.StreamHandler()
    ]
)

last_frame_time = time.time()

def tick_last_frame():
    """Simulate activity ping for watchdog."""
    global last_frame_time
    while True:
        last_frame_time = time.time()
        time.sleep(2)

def monitor_watchdog(proc):
    """If no activity in X seconds, kill ffmpeg."""
    while proc.poll() is None:
        if time.time() - last_frame_time > FRAME_TIMEOUT:
            logging.error("Watchdog timeout: killing ffmpeg")
            proc.kill()
            return
        time.sleep(1)

def monitor_ffmpeg_output(proc):
    """Scan stderr for connection errors."""
    for line in proc.stderr:
        decoded = line.decode(errors='ignore').strip()
        if "error" in decoded.lower() or "connection" in decoded.lower():
            logging.warning(f"FFmpeg log: {decoded}")
            proc.kill()
            return

def start_stream():
    """Start camera and FFmpeg pipeline."""
    global last_frame_time

    logging.info("Configuring camera...")
    camera = Picamera2()
    video_config = camera.create_video_configuration(
        main={"size": RESOLUTION},
        controls={"FrameRate": FRAMERATE}
    )
    camera.configure(video_config)
    camera.start()
    time.sleep(2)

    logging.info("Starting FFmpeg process...")
    ffmpeg_cmd = [
        'ffmpeg',
        '-fflags', '+genpts',
        '-re',
        '-f', 'h264',
        '-i', '-',  # video input from stdin (camera encoder)
        '-f', 'lavfi',
        '-i', 'anullsrc=channel_layout=stereo:sample_rate=44100',  # silent audio
        '-c:v', 'libx264',
        '-preset', 'veryfast',
        '-tune', 'zerolatency',
        '-b:v', '1200k',           # average video bitrate
        '-maxrate', '1500k',       # peak bitrate
        '-bufsize', '3000k',       # buffer size
        '-c:a', 'aac',
        '-b:a', '128k',
        '-f', 'flv',
        STREAM_URL
    ]

    ffmpeg_proc = subprocess.Popen(
        ffmpeg_cmd,
        stdin=subprocess.PIPE,
        stderr=subprocess.PIPE
    )

    # Now connect encoder output to ffmpeg stdin
    encoder = H264Encoder()
    camera.start_encoder(encoder)
    #encoder.output = ffmpeg_proc.stdin
    encoder.output = FileOutput(ffmpeg_proc.stdin)

    # Start watchdog and log monitors
    threading.Thread(target=tick_last_frame, daemon=True).start()
    threading.Thread(target=monitor_watchdog, args=(ffmpeg_proc,), daemon=True).start()
    threading.Thread(target=monitor_ffmpeg_output, args=(ffmpeg_proc,), daemon=True).start()

    logging.info("Streaming started.")
    return camera, ffmpeg_proc

def stop_stream(camera, ffmpeg_proc):
    """Stop camera and ffmpeg gracefully."""
    logging.info("Stopping stream...")
    try:
        camera.stop_encoder()
    except Exception:
        pass
    camera.close()

    try:
        ffmpeg_proc.stdin.close()
        ffmpeg_proc.wait(timeout=5)
    except Exception:
        pass
    logging.info("Stream stopped.")

if __name__ == "__main__":
    camera = None
    ffmpeg_proc = None

    while True:
        try:
            camera, ffmpeg_proc = start_stream()
            ffmpeg_proc.wait()
            logging.warning("FFmpeg process exited.")
        except Exception as e:
            logging.exception(f"Stream crashed: {e}")
        finally:
            if camera and ffmpeg_proc:
                stop_stream(camera, ffmpeg_proc)
            logging.info(f"Retrying in {BACKOFF_SECONDS} seconds...")
            time.sleep(BACKOFF_SECONDS)

```

Current stage the script can start the Picam and FFmpeg successfully but the stream broadcasting on YouTube Studio received No data.

[image]

sgbaird, 2025-04-07:
Ah, sorry - let's rename to `-mycobot280-stream` 🤦‍♂️

Too bad it didn't work out of the box. Thanks for investigating.

I think I messed up the network on the cobot stream, so I'll need to access directly unless SSH comes back somehow.

Neil-YL, 2025-04-08:
> Also, I think I unplugged and plugged back in one of the Zero 2W's, maybe the test one or the MyCobot stream, not sure.

The one on my desk is Mycobot, the one in the printer room next to the a1mini is test

Neil-YL, 2025-04-08:
> Ah, sorry - let's rename to `-mycobot280-stream` 🤦‍♂️
> 
> Too bad it didn't work out of the box. Thanks for investigating.
> 
> I think I messed up the network on the cobot stream, so I'll need to access directly unless SSH comes back somehow.

I have the hostname changed. I will continue to work on the no data issue for the new device code.

[36 more replies]