AccelerationConsortium/ac-dev-lab #164: Implement optimization algortihm operating with changing measured boundary conditions (-> recycling)

MatPopp, 2025-01-22:
#### Typical task for Bayesian optimization: 
* given 3 ingredient types (e.g. colored liquids), find mixture fractions, such that a target quantity is optimized. The cost-funciton can involve 
    * original inputs (e.g. mixture fractions, cost of materials)
    * (measured) properties of the outputs (e.g. )

#### Real-world problem: 
In many industrial processes like recycling and ore refining, the type of input / the properties of input (streams) changes over time. 
* quantity of dirt in recycled material.
* ammount of impurities in ore for refining.
* large mixing pot for a continuous process, the composition of which shall be optimized
 
Therefore, the optimized processes/mixture fractions/parameters depends on the current state of the input. One would typically perform measurements on this current state to then re-evaluate a model (describing all possible states) for the (time-dependent) optimal process parameter set. 

**This issue is meant to clarify the given problem, and to find out if/how Bayesian optimization can deal with changing input quality / state.** 

#### Idea (s): 
* perform input characterizaton measurement (for changing inputs) in each optimization iteration. (Before suggesting generating the parameter set)
* use the measured quantity as an extra input and fix it's value with a constant (equal to the measured value). This makes it necessary to re-set the boundary conditions in each iteration step @sgbaird can you point out an existing implementation of that? Looking at https://honegumi.readthedocs.io/en/latest/ the boundary conditions are typically set during experiment creation (in ax). Is it also possible to re-set boundary conditions?
* if not possible: brute-force-method: re-initialize the experiment with corresponding boundary conditions for each step, add all the previous values.

sgbaird, 2025-01-24:
Have a look at the featurization tutorial from honegumi and the Ax issues for fit_out_of_design.

Would fixed_features address what you're looking for?

For updating boundary conditions, easiest is probably to reinstantiate the client inside a loop and reattach the data

MatPopp, 2025-02-03:
@sgbaird these were the necessary hints I needed, thanks a lot! And thanks contributiong to get the necessary fixed_features in ax_client.get_next_trial implemented: 
https://github.com/facebook/Ax/issues/1951 
https://github.com/facebook/Ax/issues/905

## Minimal example: Input qualification for recycling processes with varying input quality: 

The following experiement is simulated in the code below, but can potentially be implemented as a minimal example including a physical experiment (see https://github.com/AccelerationConsortium/ac-training-lab/issues/141)

* Demo-System/Process under test: impure soap which has been diluted with water. (representing a material to be recycled, with varying, unknown composition/properties)
* Process to be optimized: (impure) soap is mixed with water, the resulting mixture shall have a defined target viscosity. 
* The "production" process is regarded as expensive (because the volume of vessels in a production plant might be too large to perform many experiments)
* the following strategy is chosen: 
    * (1) Knowledge building: Build a database of potentially occuring process parameters by actively mixing possible input materials on the (cheap) lab scale. Here: prepare many diluted water-soap inputs, characterize them (measure their viscosities), perform the optimzed process (mix them with pure water), measure resulting viscosities 
     * (2)  Production optimization: Make an input qualification measurement (here the viscosity of the impure soap-input). Use this as a fixed feature for further optimizaiton (in order to save iterations optimizing the expensive process)

for images below: target viscosity = 10 (mPas). 
Model slice at "measured" input viscosity, before performing "production experiments": 

[image]

after only 4 iterations a viscosity of 9.34 (mPas) is reached

[image]

the best value after 10 iterations is 10.026 (mPas)

[image]

Looks promising to me but thorough statistical testing remains to be done. 

(very preliminary) code for (simulation) experiment above:
```python
## based on https://honegumi.readthedocs.io/en/latest/
## target of this script: use fixed but changing features in the optimization, representing varying input-quality of
# recycling / refining processes
# the idea is to use the fixed_features argument of the AxClient

import numpy as np
import pandas as pd
from ax.service.ax_client import AxClient, ObjectiveProperties
from ax.service.utils.instantiation import FixedFeatures

from ax.modelbridge.factory import Models
from ax.modelbridge.generation_strategy import GenerationStep, GenerationStrategy

from ax.plot.slice import plot_slice
from ax.utils.notebook.plotting import render

from botorch.acquisition import UpperConfidenceBound
from pydantic import BaseModel, Field
from typing import Optional

import json

import time

## creating some mock-up funcitons:

## models for simulation purposes. Not known by the optimization algorithm
def soap_mix_viscosity( soap_fraction, soap_purity = 1):
    ## 1.0 for pure water, , 2000 for pure soap, increasing in between, cubic estimation.

    actual_soap_fraction = soap_fraction * soap_purity
    ret = 1.0 + 2000.0 * actual_soap_fraction ** 3
    return ret

# Perform model optimization part:
# The Optimizer is allowed to synthesize impure input solutions first, such that it can use the generated knwoledge
# later for faster optimization of the target function.

[... 165 lines cut ...]

    ## document the experiment
    experiment_series.experiments.append(SoapExperiment(
        observed_soap_input_viscosity = input_viscosity,
        impure_soap_fraction = impure_soap_fraction,
        mix_viscosity = measured_output_viscosity
    ))

    ## some visualization
    model = finetune_ax_client.generation_strategy.model
    render(plot_slice(model, param_name = "impure_soap_fraction" ,metric_name = "target_deviation",
                   slice_values={"input_soap_viscosity": input_viscosity}))

best_of_all_result = finetune_ax_client.get_best_parameters()
best_trial_index = finetune_ax_client.get_trials_data_frame()["target_deviation"][n_explore::].idxmin()
print("best of all result:", best_of_all_result, "watch out, this can be out of the pre-experiments")
print("best trial index:", best_trial_index)
print("experiment data of best trial index:", experiment_series.experiments[best_trial_index])
#print(json.dumps(experiment_series.dict(), indent=4))

## TODO: comparison to the following cases:
## 1. no input characterization, use previous soap model, ignoring that soap is impure
## 2. no input characterization, start optimization from scratch

```