AccelerationConsortium/ac-dev-lab #168: A1 Mini 3D printer - Self-driving Lab for MSE403H1 practical

sgbaird, 2025-01-23:
Students in the course will send commands (likely: nozzle temperature, printer speed, extrusion rate, and bed temperature) to a [Bambu Lab A1 Mini 3D printer](https://bambulab.com/en/a1-mini) to have a small, single-layer object printed (likely a small square or a simple logo/icon) after which point an overhead camera will take an image. Print quality metrics (likely: maximize extrusion overlap, minimize extrusion mismatch, minimize cost) will be calculated and returned to the students. The job of each student is to find the optimal combination of parameters that optimizes quality. The students will interact with the system in a "gray-box" format. The students will be provided a Colab notebook that contains a function to send commands, receive the image, and calculate the quality metrics. It will be up to them to choose/design an optimization strategy.

On our side (inside the gray box), the workflow will looks somewhat like the following (diagram pulled from https://github.com/AccelerationConsortium/ac-training-lab/issues/160#issuecomment-2610916287), with the following key related issues:
- #149 
- #147 
- #158 
- #159 
- #160 
- #161 

> [image]
> 
> (if you have write access to this repo, you should be able to [view and edit the diagram directly](https://app.diagrams.net/#HAccelerationConsortium%2Fac-training-lab%2Fdrawio%2Freports%2Ffigures%2Fdrawio%2Fa1mini-sdl.drawio#%7B%22pageId%22%3A%22oiAiHZ9bc04vitCAztIp%22%7D))

SissiFeng, 2025-02-10:
https://utoronto-my.sharepoint.com/:w:/g/personal/sissi_feng_utoronto_ca/Efo8kI05SrRIsMQshBLWC8sBrom8XhHEND6Th2kAihnGfQ

sgbaird, 2025-02-12:
- to prevent stringing ruining later iterations, constrain the movement so the extruder is never passing over unused slots (if it strings onto old squares that already had the picture taken, not a big deal)
- I'm not as worried about 10 min per iteration, better than not being implemented. Let's stick with single GCode file per square probably.
- please implement a purge / reset step at the end of each iteration. Choose some standard set of parameters (maybe the default recommended ones), go to a position off the build plate, extrude for some amount of time (e.g., 1-2 min) onto the desk
- at the *beginning* of each step, do the same: move to waste extrusion position, change to the new settings, let everything get into a steady state for some amount of time (e.g., 5 minutes), extrude for a minute or two, then carry out the iteration. Again, make sure that unused squares aren't "trespassed" until it's time to actually use them
- typically, movement should be far up in the z-direction before moving in XY in-between iterations. XY movement into other squares with low clearance between build plate and extruder is more likely to cause issues. I.e., lift the extruder up all the way before moving it in XY when switching slots, etc.
- material build-up on nozzle. Thanks for the picture - that helped me understand. Maybe requires a removal stand. Something really simple. See video below: 

https://github.com/user-attachments/assets/996586d2-3c6a-4b80-aa55-901cbfad2d31

Otherwise, we can try to position the livestream camera so that we can see if there is material build up on the extruder nozzle. If there are certain conditions that are more likely to produce that build up, we can try to narrow what's allowed (e.g., not letting nozzle get too hot)

Keep the other parameters as possibilities for people to adjust (fan speed, print speed, etc.). Something to be left to the students. They get to figure out what's important or not.

SissiFeng, 2025-02-12:
Nozzle buildup needs to be heated to 220-240 for easy removal, so the stand needs to consider that temperature.

SissiFeng, 2025-02-12:
We dont update the firmware, but today sending print jobs directly to the Bambu A1 Mini using Python has stopped working. The printer fails to parse the file and displays the following error message:

[image]

Neil-YL, 2025-02-12:
> We dont update the firmware, but today sending print jobs directly to the Bambu A1 Mini using Python has stopped working. The printer fails to parse the file and displays the following error message:
> 
> <img alt="Image" width="342" src="https://private-user-images.githubusercontent.com/38707787/412529782-f6745bfb-28c4-4141-a9b8-1478381cb11b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkzODU0MjcsIm5iZiI6MTczOTM4NTEyNywicGF0aCI6Ii8zODcwNzc4Ny80MTI1Mjk3ODItZjY3NDViZmItMjhjNC00MTQxLWE5YjgtMTQ3ODM4MWNiMTFiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjEyVDE4MzIwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJmZTg2ZmQyMTZhODRkYmQwZGQyNTE1OTdkMWY0ZDg0YjFiN2U3MTcxOWIxNWY5Nzg3YWZiZDJhMWNhNmNlYmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.lJ12XT--pcYho6O7YPijt7apst5IFRDUzbu_9aIOQQ8">

Previously, I successfully sent the same G-code file to the printer and had it printed without any errors. And Sissi witnessed the process. 

However, today I encountered the same issue above, although my error code is slightly different: 0500-4003 108000. I have tried using another G-code file (exported directly from Bambu Studio), setting up a new environment for the BambuAPI, and restarting the printer, but the issue persists.

Neil-YL, 2025-02-12:
Some advice from the community forum said the original SD card comes with the printer is poor quality so I tried with a new SD card still received the same error:

[image]

sgbaird, 2025-02-14:
Could you open an issue on the bambu lab API package repo? (Assuming one isn't already there)

sgbaird, 2025-02-14:
> Nozzle buildup needs to be heated to 220-240 for easy removal, so the stand needs to consider that temperature.
> 
> Another scenario to consider: Let's say the current parameter settings are 210/70 (the recommended default optimal parameter is 220/65). Instead of spitting out the normal amount and sticking to the bed, the filament is carried by the nozzle head with the spit out portion on the plate to complete the whole movement, and finally this time the task camera has no square to capture. This part of the filament stuck to the nozzle head is not sure if it will be completely removed in the reset step at the end.

Could you draw/include a picture to help me understand? Would having a small "trash" area to the side of each square, where the reset procedure begins, help with this? If we can get a reasonable reset procedure going, maybe this issue can be tested and kept in mind. Is this something that you think would happen frequently?

Neil-YL, 2025-02-14:
> Could you open an issue on the bambu lab API package repo? (Assuming one isn't already there)

I updated the new problem to our previous issue

SissiFeng, 2025-02-14:
(The video uses a continuous printout for illustrative purposes, not the single square printout that will be used for the experiment.)

1. Reset and “Trash” Wipe at the Start of Each Print
At the beginning of every print task, there is already a reset procedure where any leftover filament from the previous job is wiped off onto the lower edge of the build plate (as seen around the 1:35 mark in the video). A similar wiping action occurs again around the 3:15 mark in the video, removing any excess filament from the nozzle head.

2. Non-Optimal Parameter Settings Can Cause Poor Adhesion
We experimented with parameters of 210°C nozzle temperature and 70°C bed temperature (instead of the recommended 220°C/65°C). In the video at about 4:24, you can see a situation where the filament does not properly adhere to the bed. As a result, part—or sometimes all—of the printed square remains stuck to the nozzle head rather than the build plate. If the entire piece is lifted and carried away by the nozzle, then by the time the print finishes, there is no square left on the build plate for the camera to capture.

How frequency:
We have conducted multiple print trials using the 210°C/70°C settings. The print quality has been inconsistent:
sometimes the square prints normally.
sometimes a corner is missing.
sometimes the entire piece is gone.
We have not yet recorded exact statistics on how frequently each issue occurs.

3. Temperature Drift
Another observation is temperature fluctuation. At around 3:28 in the video, the set temperature is 220°C, but the actual nozzle temperature drifts among 222°C, 221°C, 220°C, and 219°C. This slight fluctuation could also play a role in inconsistent print adhesion.

sgbaird, 2025-02-14:
Thank you for the video and description! Very informative.

sgbaird, 2025-02-15:
I think it's ok if the piece gets lifted off as long as it doesn't interfere with later prints. This can essentially be considered a failure condition (I.e., in a real print, if that happens it would probably be unrecoverable). We need to make sure the algorithm that computes metrics can handle this. Maybe we need some assessment of color of the filament. But we can worry about that a bit later. It's ok if there's some inconsistency, as long as we've done our best to make sure a proper reset has taken place. Position on the plate will affect quality, but as long as we tell people the next position available on the plate / which position had been printed on, I think we're ok.

sgbaird, 2025-02-25:
I moved a lot of the discussion about the print workflow to https://github.com/AccelerationConsortium/ac-training-lab/issues/156#issuecomment-2683252975.

sgbaird, 2025-02-25:
@Neil-YL could you also work on switching to one of the AccelerationConsortium clusters for the MongoDB piece? Looks like "Live migration" might be the way to do it https://www.mongodb.com/community/forums/t/migrating-entire-cluster-to-different-mongodb-account/172698. I can make sure you have whatever permissions you need on the org.

Neil-YL, 2025-02-26:
Sure.

sgbaird, 2025-03-04:
Adding https://github.com/AccelerationConsortium/ac-training-lab/issues/179 for more in-depth discussion/update around the database integration. Related to #159 and #160 especially, and following a similar setup as OT-2, which will be have some additional documentation in this PR: https://github.com/AccelerationConsortium/ac-training-lab/pull/170#issuecomment-2683294323.

sgbaird, 2025-03-10:
Planning to come in tomorrow to work together on the project