AccelerationConsortium/ac-dev-lab #172: Inventory Database Table

RubainaFarin31, 2025-01-24:
Inventory Database Table Plan: January 23rd:

- Potential two ways to go about it:
a) Develop a database using PostRes or MongoDB -> hosting using AWS
b) Create integration with existing Airtable 

Currently, the Airtable option seems to be the most user friendly and accessible, as the AC has used Airtable in the past and integration will be easier for users. Should the airtable option be used:
- Find ways to integrate AprilTags to update Airtable inventory
- Look into integration methods for Airtables

---
General plan:
1. Explore both options -> see which is better in terms of long term usage, storage and user design
2. After choosing options, begin developing a database
3. Look into AprilTag integration and automated updated

sgbaird, 2025-02-15:
Checking back on a couple projects. Could you provide an update here?

RubainaFarin31, 2025-02-20:
> Checking back on a couple projects. Could you provide an update here?

Hey Sterling — hope all is well! Will and I met a couple of weeks ago to discuss the project. I had proposed the idea of an Airtable integration. He had mentioned that the AC is looking into another separate inventory system (SciNote I believe), which he wanted to explore first. Should that not meet our needs, we can circle back to create the integrated system. 

He had suggested that regardless of the AC using this, I should go ahead and work on this as a side project, given the usefulness of the proposed project.

sgbaird, 2025-02-26:
Some of the fields: PO number, order date, quote #, vendor, product number, # of items, order for, SDL/office, category, total cost, received, date received, notes, ..., item link, ...

An example of an entry:

[image]
[image]
[image]
[image]
[image]

Could you open a post on the Accelerated Discovery forum asking about how people are approaching tracking their inventory (especially with autonomous systems in mind)? I think it would be well worth doing some crowd sourcing of ideas.

RubainaFarin31, 2025-02-27:
Thanks for this Sterling! Just added a post on the Discovery forum: https://accelerated-discovery.org/t/approaches-to-inventory-tracking-with-autonomous-systems/477

Will keep track of this and keep you updated!

sgbaird, 2025-03-22:
Any updates here?