AccelerationConsortium/ac-dev-lab #174: Automating Pioreactor workflow to create a continous experiment

linx5o, 2025-01-27:
We aim to implement functionality and procedures that allow for continuous experimentation on the Pioreactor. We will maintain a human-in-the-loop approach to unnecessary engineering.

TODO:

- Talk to industry experts to understand the uses of the Pioreactor and how it could be used.
- Implement workflows to allow for constant experimentation.
- Allow for Pioreactor to call for help, through notification.

sgbaird, 2025-02-15:
Checking back on a couple projects. Could you provide an update here?

linx5o, 2025-02-16:
I have developed a workflow with Prefect. This would allow us to control all aspects of the pioreactor via the Gradio API. I have also configured a self hosted docker based Prefect server. The flow can be set to have pauses and await for human input. 

This approach would require the customized flows for specific use cases. More generic flows can also be used. (For example, turning on and off the stirring.

sgbaird, 2025-02-28:
> I have developed a workflow with Prefect

Where can I find this?

> I have also configured a self hosted docker based Prefect server

Nice! Could you share more about this?

> The flow can be set to have pauses and await for human input.

Not urgent, but it would be great to have a video demo of this

sgbaird, 2025-02-28:
Any updates on the conversations with SDL6?

linx5o, 2025-02-28:
> Any updates on the conversations with SDL6?

I am currently working with scientists at SDL6 on API to integrate some of their devices. They require sterile conditions as well as temperature, humidity, oxygen, etc. We can't really control all of these with the Pioreactor.

linx5o, 2025-02-28:
> Where can I find this?

I put it in the same branch as the pinch valve. relay-support-for-pioreactor. Only a simple stirring flow right now. Does support inputs though. Currently there are no notifications for the stop. But it would be a simple choice of choosing what to notify on and adding it before pausing.

> Nice! Could you share more about this?

 I have a truenas scale server running at home. I am using portainer (a docker container manager) to deploy a prefect server and a worker. This was after deploying the workflows we can use them remotely through the web UI without having to trigger the scripts again. I have put the compose file for the server in the prefect subfolder of the pioreactor folder on training lab repo (same branch).

> Not urgent, but it would be great to have a video demo of this

OK

sgbaird, 2025-02-28:
> I am currently working with scientists at SDL6 on API to integrate some of their devices. They require sterile conditions as well as temperature, humidity, oxygen, etc. We can't really control all of these with the Pioreactor.

Thanks! That sounds great. Are you pushing the code to repos on the SDL6 gh organization? I'd like to know which devices and the general approach (e.g., which communication protocols, level of existing programmability of the devices).

Agreed about there being limitations with the Pioreactor. A few points:
1. From pioreactor docs on [sterilization](https://docs.pioreactor.com/user-guide/prepare-vial-for-cultures#3-sterilization):
> The glassware vial, stir bar and vial cap that come with your Pioreactor are all fully autoclavable, and heat resistant up to 125℃. The luer lock ends can be covered with tin foil, or attached to air filters, or neither.
[Also:](https://pioreactor.com/en-ca/products/temperature-pro-kit)
> 47mm Stainless Steel Thermo-well with G1/8 Threads: engineered to be fully autoclavable, guaranteeing sterility.
Though I know there are some pretty stringent requirements for SDL6
1. For temperature, is the idea that the upper limit of the PCB heating + high-T expansion isn't sufficient (~45 C)? Seems like [with some trial-and-error and an incubator, 60 C is within reason](https://forum.pioreactor.com/t/stirring-malfunction-at-high-temperatures/349/23). Do you know what they'd need in terms of an upper limit on temperature?
2. If we're talking about chamber humidity, an incubator would also help
3. For oxygen (and vial humidity), is the issue being able to precisely control flowrate (i.e., with the pinch valve)?
If a certain mixture of oxygen and humidity (and e.g., CO2) is required, this could be addressed by having multiple gas lines going to a manifold before entering the vial. With four pinch valves: [`dry O2`, `wet O2`, `dry CO2`, `wet CO2`], a mixture of O2, water vapor (up to the % in the wet gases), and CO2 could be achieved.

Aside: could you follow up at [pioreactor forum](https://forum.pioreactor.com/t/pinch-valve-connector/631) or open a new topic asking about if the pinch valve can be partially open, in case they're already aware? I suppose at the point where controlling flowrate is important, a [flow controller](https://www.takasago-fluidics.com/products/controller-pumps-valves) of some kind is necessary..

Very well it might be the case that Pioreactor can't be used directly in SDL6's priority workflows; however, there's a large value to knowing what the gap / limitations are (i.e., characterizing Pioreactor on the spectrum of cost-fidelity in the context of real-world workflows).

Thanks for these updates!

linx5o, 2025-03-01:
> Thanks! That sounds great. Are you pushing the code to repos on the SDL6 gh organization? I'd like to know which devices and the general approach (e.g., which communication protocols, level of existing programmability of the devices).

I am pushing code to the SDL6 Repos. There are many devices to be worked on. Currently working on a [centrifuge](https://www.agilent.com/en/product/automated-liquid-handling/automated-microplate-management/microplate-centrifuge), the company has not released any APIs or driver, we have turned to a community-developed driver from [Pylabrobot](https://docs.pylabrobot.org/user_guide/centrifuge.html). It uses a COM connection to the device. Looking through their code, I see they have reverse-engineered commands and are sending raw binary. The end goal is to integrate an API into Prefect so that we can create workflows and integrate with other devices.

> Agreed about there being limitations with the Pioreactor. A few points:

The main concern is that the 15ml vial is too large. They want to use 96 Well Plates (I guess for cost reasons, not sure though). They asked about the possibility of passing a tube through the reactor and having it act as a measurement device (or having the experiment take place in the tube due to smaller volume), but I think this would be infeasible as the tube would make the sensors inaccurate. 

For temperature I think this might be the only thing that meets their requirements. 

For atmosphere, they want to control the amount of oxygen and CO2 in the environment and be able to measure and change according to changes (being consumed). I think they wanted to focus on the variations in these amounts. I am not familiar with how to do this.

[Partially open pinch valve](https://forum.pioreactor.com/t/pinch-value-partially-open/682)

sgbaird, 2025-03-01:
> I am pushing code to the SDL6 Repos. There are many devices to be worked on. Currently working on a [...]

Cool! Could you include the links to the repos you're working on? It's OK if they're private - I have access and makes it easier to navigate to.

> The main concern is that the 15ml vial is too large

Ah, that makes sense! So, ~50x larger than what they're hoping for 😅 (96-well plates typically 200-300 uL). Not sure there's a great way around this in the Pioreactor system, though maybe some of the hardware could be reused. Non-trivial for sure. That's good to know the temperature is an appealing piece to the system (if I understood correctly). For atmosphere, measuring consumption is where things _might_ get expensive, depending on how accurately they want to measure. Though if I'm thinking about this correctly, I imagine with good optical density values, they might not need to care as much about consumption.

linx5o, 2025-03-03:
[Vspin Driver](https://github.com/AC-SDL6/VSpin_Driver) - This is the current one. Mostly just troubleshooting and documenting the user of external driver.

> Ah, that makes sense! So, ~50x larger than what they're hoping for 😅 (96-well plates typically 200-300 uL). Not sure there's a great way around this in the Pioreactor system, though maybe some of the hardware could be reused. Non-trivial for sure. Eventually would be good to check if it's a cost thing (reagents), an integration thing (wellplates being part of the common workflow) or a scalability thing (many pioreactors == $$, more space required, etc.), or something else. That's good to know the temperature is an appealing piece to the system (if I understood correctly). For atmosphere, measuring consumption is where things might get expensive, depending on how accurately they want to measure. Though if I'm thinking about this correctly, I imagine with good optical density values, they might not need to care as much about measuring consumption via gas composition changes.

Yep

linx5o, 2025-03-03:
Partial open solenoid:

"Probably not - most pinch valves work through a solenoid, which is very much a on / off tool. You’ll need to more advanced set up if you want a three-state valve."

From [Forum](https://forum.pioreactor.com/t/pinch-value-partially-open/682/2)