AccelerationConsortium/ac-dev-lab #176: Potential of using AS7341 to measure NIR absorption for MSE403H1 Multi Objective practical

Neil-YL, 2025-02-19:
Here is a quick note of my test and findings for today:

I modified the code on an idle AS7341 color sensor package, adding NIR results output (Now it returns 8+NIR channel intensity data)

I measured followings samples at well B2, D2, F2, with current LCM Practical environment (Room light off, OT-2 light on and sensor LED off):

- Blank well (with a white A4 paper at the bottom of the plate)
- White acrylic paint liquid
- Red acrylic paint liquid
- Yellow acrylic paint liquid
- Blue acrylic paint liquid

WRYB / Blank: Normalization relative to the blank sample.
RYB / White: Normalization relative to the white sample.

[image]

Some findings:

- White is very closed to blank. We can use white paper background as reference later, no need to prepare white paints.
- Yellow is very similar with Red in 620, 670 and NIR, using 620, 670 may not enough to evaluate **Redness** of a color.
- Yellow even has a higher reflectance than red. I suspect it is becasue yellow is more transparent than other so the reflection by white background contribute to the measurement.
- NIR absorption of blue is not very significant in value, (relative reflection around 83-84%, ～16% absorption). Not sure how much this would allow us to add glucose to achieve lower reflectance and higher absorption, but at lease it is not overwhelmingly 50% absorption. I haven’t delved into the literature yet, but from a quick search, a 5% glucose concentration has about 10-12% NIR absorption, while a 10% concentration has approximately 16%

Some concerns:

-  absorption would be A = 1 - (R+T). R: Reflectance; T:  Transmittance. We are using white paper to convert all T into R in this measurement. I am not quite sure if this could a problem (says adding solute to the liquid reducing the transmittance of the liquid so less T but we will take it as more A)
- Methods to evaluation Redness of the sample.

Sissi will also comment with her work from the experimental design perspective and algorithm for the evaluation later.

I may not be able to run more tests before Monday, as students will be using the OT-2 extensively for their assignments on Thursday and Friday. However, any further experimental suggestions or discussions are very welcome.

Neil-YL, 2025-02-19:
Data set: 

[NIR_Measurement_Results.csv](https://github.com/user-attachments/files/18876094/NIR_Measurement_Results.csv)

SissiFeng, 2025-02-19:
Current Test Data & Results：

(1) Direct analysis of channel intensities

[image]

ch620 vs. ch670 (red channel correlation):

The results show that ch620 and ch670 have a very high correlation (close to 1.00), indicating that they can be used together for red detection.

NIR vs. Red channel (NIR effect on ch620 & ch670):

NIR was found to have a high correlation (>0.92) with the red channel, but the trends were not entirely consistent.

Channel intensity distribution (ch620, ch670, NIR for different samples):

The ch620 and ch670 of the red sample are significantly higher than the other colors, but the NIR channel does not show the same pattern.

The blue sample has a significantly lower response on NIR and is clearly distinguishable from the other colors.

NIR/ch620 ratio analysis:

The NIR/ch620 of the blue sample is significantly higher than that of the other colors, indicating that the ratio can be used for color classification.

(2) Ratio analysis method

[image]

Calculate key ratios (620/470, 670/470, NIR/620, etc.):

Red vs. Yellow has a strong categorization ability, and 620/470 and 670/470 form a clearer distribution of categories.

The NIR/620 ratio is effective in distinguishing blue samples and remains stable in samples with low NIR uptake.

Correlation analysis (ratio vs. color channel):

The correlation between 620/470 and 670/470 is extremely high (>0.99), suggesting that the two can be used together for red color identification.

NIR/620 correlates negatively with 620/470 (-0.89), indicating that NIR transmission is lower when red absorption is higher.

The lower correlation (0.04) for the 470/440 channel indicates that the spectral behavior of blue and yellow is fundamentally different from that of the red channel.

3 Main findings

Direct use of channel intensities (ch620 & ch670) can distinguish red color well, but not easily distinguish yellow color.
The use of ratios (620/470, 670/470, NIR/620) further improves the classification ability, especially in distinguishing yellow from red.
The NIR absorption pattern is different from the red absorption pattern, and the NIR/620 ratio is better at distinguishing blue samples.
NIR/620 is negatively correlated with 620/470, suggesting that both may need to be optimized separately in the MO objective.

Our thoughts:
The lower blue absorption may make it easier to observe the effect of sugars on NIR.
If the relative increase in NIR uptake is ≥50% with the addition of glucose, then it can be assumed that sugars significantly affect IR uptake.
In our next experiment, we would like to try different sugar (glucose, fructose, sucrose) concentrations to observe the linkage between NIR and visible channel.

Neil-YL, 2025-02-24:
Measured 10% glucose and 20% glucose blue/ yellow paint liquid today. Differences for each channel are not very distinguishable comparing with no glucose blue/yellow paint:

(Measured results converted into normalized relative reflectance as above)

[image]

[image]

[image]

[image]

[Glucose_NIR_Measurement_Results.csv](https://github.com/user-attachments/files/18949400/Glucose_NIR_Measurement_Results.csv)

Glucose has specific absorbance spectrum in NIR range, 850nm–2500nm, however, the measuring spectrum for AS7341 only covers up to 1000 nm. This could explain why the difference is not very noticeable in our measurements.

closed 2025-02-26 by Neil-YL