AccelerationConsortium/ac-dev-lab #181: SDL0 bimanual robot project

gursi26, 2025-03-07:
This issue is just a summary of what I have been working on at SDL0, as requested by @sgbaird in #74.

So far, we are working on setting up a bimanual robot setup with two UR5 arms and zed depth cameras mounted on the robot wrists and around the scene for accurate depth estimation.

we hope to use it for [3dgs](https://repo-sam.inria.fr/fungraph/3d-gaussian-splatting/)/[nerf](https://www.matthewtancik.com/nerf) reconstructions of the scene and leverage those for accurate robot manipulation and interaction with objects in the scene.

sgbaird, 2025-03-22:
Any updates on how this one is going?

gursi26, 2025-03-23:
yep, just this week we finished building the setup.

we have set up two UR5e arms with robotiq grippers on the front. we also setup 4 zed mini depth cameras, with two mounted to the robot arm with a 3d printed attachment made by Kelvin. The other 2 depth cameras are mounted on camera stands that we clamped to the table, so they can look down on the scene. 

since the setup is complete we are probably going to move on to calibration of the cameras along with testing the setup out with some simple pick and place tasks.