AccelerationConsortium/ac-dev-lab #216: vertical lift module - vibration testing

sgbaird, 2025-04-02:
Using [Reed R7000SD](https://www.grainger.ca/en/product/VIBRATION-METER/p/REER7000SD) (also needed to purchase bought https://www.amazon.ca/dp/B004ZIENBA, and bought https://www.amazon.ca/dp/B09Z6JCKL7 just in case the person who's doing the testing doesn't have the right adapter)

sgbaird, 2025-05-22:
Aside: UPS that Ilya from SDL6 has, APC

[image]

sgbaird, 2025-09-04:
We have some data and videos here courtesy of Nick Wilk from Johnstone Equipment that I'll upload.

sgbaird, 2025-09-15:
@pranshu-ui I'd like you to follow-up with Nick to schedule time to bring a microscope and a UPS with you to test out. This will require some setup in advance. The idea is to have a microscope on a tray recording the live feed, the microscope powered by a UPS, a vibration meter collecting vibrational data, and an external camera taking a video of the full setup. The test workflow will be the same as above, and in the end we'll have a comprehensive set of data.

pranshu-ui, 2025-09-25:
Going to do vibration testing on Monday at 1PM at the JohnStone facility.

sgbaird, 2025-09-26:
For the microscope, perhaps the best starting point is https://openflexure.org/, watching the 2-min YT video and jumping to ["use the microscope" for the v7 microscope](https://openflexure.org/projects/microscope/control). The GUI is already installed, and I think there's a desktop icon.

In addition to the official documentation, there are lots of discussions on this repo or the AC forum related to its use at the AC:
- https://github.com/AccelerationConsortium/ac-dev-lab/issues?q=is%3Aissue%20openflexure
- https://accelerated-discovery.org/t/building-the-high-resolution-motorized-openflexure-microscope-v7-using-the-rodeostat-kit/231
- https://accelerated-discovery.org/t/building-the-open-flexure-delta-stage-reflection-and-transmission-microscopes/254

See also https://openflexure.discourse.group/t/accessing-the-raspberry-pi-camera-openflexure-connect-video-stream/1706/6

I think there was a script @kenzo-aspuru-takata had saved somewhere that did the streaming, and we need a new livestream URL of course. However, it might be simpler and more robust in the meantime to have it record to disk. I'll open a separate issue.

kenzo-aspuru-takata, 2025-09-27:
That livestream script was kind of a temporary solution. It was basically a macro that moved the mouse around and clicked on the screen, and then ran a console command to start a livestream. Rather than trying to find it or something, it would probably be better to rewrite that code or use that disk solution. I don't know where that code is anymore, either.

pranshu-ui, 2025-09-30:
### Went in today. Collected valuable information about the VLM.

**The Setup**
- Openflexure Microscope
- UPS operating on battery
- Phone stand with phone recording & providing light
- A laptop streaming microscope feed
- Vibration Meter

**Microscope Setup (removed once setup)**
- Monitor
- USB-A keyboard and mouse
- hdmi to microhdmi cord

[image]

**Key Notes**
- The bottom of the tray, like this one, can be lined with a ferromagnetic sheet to secure components. In this case the vibration meter's recipticle was magnetic, which made testing more reliable.
- The VLM vibrates with the most intensity when going in and out of the open port, not really when going up or down. 
- The speed of the VLM can be reduced to reduce vibration by the manufacturer upon order. Many companies, especially in research/aerospace tend to do this.
- The microscope itself, despite the vibrations, did not falter in its focus. Although the sample shook, it was very very fine and was still  usable even while it was in the VLM.
- The VLM tray was large enough as to not have other heavier loads effect its vibrations. Although lighter loads must be secured due to rapid accelerationa and decelerating motions.
- We repeated the test 3 times. Data is attached.

https://github.com/user-attachments/assets/ad21bfd0-be16-47a3-b565-75c9eb7ddfc6

Data:

sgbaird, 2025-10-02:
It sounds like something went wrong with the screen recording earlier. Could you set up a new time to meet with Nick and collect the full dataset (video of the tray, vibration data, and video feed of the microscope) that we can correlate via timestamps and show altogether? (Picture a ppt slideshow where all three are being shown simultaneously, sync'd to the same timestamps)

sgbaird, 2025-11-03:
VLM no longer being ordered. Closing for now. May revisit later

closed 2025-11-03 by sgbaird