AccelerationConsortium/ac-dev-lab #229: Create AprilTag minimal working example

sgbaird, 2025-04-17:
Likely with the point and shoot RPi camera (see issue) which also has the benefit of the ring light compatibility.

sgbaird, 2025-05-22:
Aside: @yakavetsiv using AprilTags

[image]

Generally tries to use two AprilTags to pose ambiguity

Planned use with depth camera that SDL3(?) also has. Custom drilled/installed onto plate crane

[image]

[image]

Doing something interesting with trying to avoid calibration (maybe didn't fully understand, but I think had to do with having a calibration AprilTag that's fixtured to the starting local reference frame?)

ZZZhaoz, 2025-06-02:
I’ve got a few ideas for the program. Let me know if they make sense!

## Program Proposal

Since AprilTags can only be attached to the side of the turntable, I propose fixing the camera to the base of the MyCobot, facing the turntable horizontally.

### Workflow

1. The turntable rotates slowly after startup.
2. The camera continuously takes photos to detect AprilTags. The camera position or focus can be adjusted to make sure only one or a few tags appear in each frame.
3. Once the target AprilTag is detected, the turntable stops. The corresponding vial is now in a good position for the robot to grab.
4. The system re-detects the tag and converts its position from image coordinates to robot coordinates using a pre-calibrated transformation matrix. The MyCobot then grabs or places the vial from the side.
5. Users can optionally control the rotation speed to increase uncertainty in the stopping position, making the use of AprilTags' position estimation necessary and meaningful.

### User Experience

Since we detect AprilTags on the turntable, not on the vials themselves, we can’t directly tell which vial is in which slot. To solve this:

- Each vial gets a physical label (Vial ID).
- Each position on the turntable gets a physical label (Turntable ID) (e.g., placed at the center area).

We maintain two bidirectional mappings in the program:

1. Turntable ID ↔ AprilTag ID  
2. Turntable ID ↔ Vial ID  

This way, the user can say something like “place vial 3 at position 5,” and the system can automatically figure out which AprilTag to look for to pick up the vial and which one to use to place the vial.

ZZZhaoz, 2025-06-03:
Updated: [apriltag_sheet (38).pdf](https://github.com/user-attachments/files/20577085/apriltag_sheet.38.pdf)

ZZZhaoz, 2025-06-09:
Just following up on the hardware setup. Are there any updates? I'm still waiting to start development. Thanks!

sgbaird, 2025-06-11:
See https://github.com/AccelerationConsortium/ac-training-lab/issues/270#issuecomment-2961203002 (sorry, some of that probably should have gone to this issue instead)

ZZZhaoz, 2025-06-11:
Okay, I’ll give the cobot a try and let you know how it goes.

sgbaird, 2025-07-10:
> Because the cobot has a limited range

Very clear and thorough implementation. I can appreciate where this took so much debugging. The minimal working example requires a three-system deployment (hugging face, colab, and the device), along with the matrix issues you mentioned, the cobot hardware/software instability, refining / making more robust the communication between HF and device, and the very limited reach/flexibility for the intended poses (partly hardware and partly their coord and inverse kinematics calculations).

It seems to really struggle with grasping from the side while using coords, and I've wondered if using gradient descent with angles directly might help.

Likewise, I'd be ok with grasping from the top of the vials, too. We could either use the tags on the vials directly or still take an image from the side and make the longer trip to pick from the top. This "grab from top" approach matches some of the setups in other labs. They also have some custom grippers with four prongs.

Aside: today I also resolved a few of the communication issues with being unable to query angles and switching over to using sync_send_angles, etc.

I kept having trouble with it responding to the request to "move down". Do you mind reaching out to elephant robotics folks' support email explaining what's happening, with me cc'd?

I was able to verify basically everything except the "move down". I have a Colab copy in my Google drive that needs to be uploaded which has a few minor edits. This will also need to be merged with any changes / PRs you made today (sorry).

---

For the token, the plan is to put the cobot in an environment where it's not a problem for it to capture images (e.g., putting some sheets around it so it doesn't capture people's faces) and make it public. Alternatively, we make it public and use some kind of API system similar to https://huggingface.co/spaces/AccelerationConsortium/OT-2-LCM. Likely, this would be more of a login key rather than credits to be consumed, as in the OT-2 demo.

---

Aside: for the rotation and translation matrix operation fixes, does anything need to be adjusted on the first "self-contained MWE" camera calibration notebook?

zweaung1014, 2025-07-15:
@ZZZhaoz Hey Jason, we were trying to run the ur3 robot, had some questions on the hand-eye calibration code on there. 

We are following this notebook: [Notebook link](https://colab.research.google.com/drive/18-WF_GoNFaZXC0TZAXvvQPDhPcTIvZFL?usp=sharing#scrollTo=f13dbd57)
We see the two offsets for  pick-up and placement here, but **where is the hand-eye offset (the offset between the camera and the gripper) defined?**

Also, I'm not very clear about how `hand_eye_calibration.npy` works:
[image]
The current hand_eye_calibration.npy the notebook uses is for the camera on the cobot. **How should I go about getting this for the camera that's on the ur3 robot?** 

Do I need to follow this?
[Notebook link](https://github.com/AccelerationConsortium/ac-training-lab/blob/apriltag_cobot_example/src/ac_training_lab/apriltag_hardware_demo/hand_eye_calibration.ipynb)
I also have a few question on this: 
[image]
If yes, **how is camera_params.npy different from hand_eye_calibration.npy?**

ZZZhaoz, 2025-07-16:
Yes, you need to follow [Hand-Eye Calibration](https://github.com/AccelerationConsortium/ac-training-lab/blob/apriltag_cobot_example/src/ac_training_lab/apriltag_hardware_demo/hand_eye_calibration.ipynb) to obtain the specific camera intrinsics and the hand-eye calibration result for the ur3 robot. camera_params.npy contains the camera intrinsics, including focal length, principal point, and distortion coefficients. hand_eye_calibration.npy contains a 4×4 matrix describing the relative transformation between the camera and the cobot’s end-effector. These parameters are crucial for accurate position estimation and  precise grasping.

The notebooks were developed for the cobot, so all offsets in [apriltag_hardware_demo.ipynb](https://colab.research.google.com/drive/18-WF_GoNFaZXC0TZAXvvQPDhPcTIvZFL?usp=sharing#scrollTo=f13dbd57) should be adjusted to align with the ur3 robot. For [Hand-Eye Calibration](https://github.com/AccelerationConsortium/ac-training-lab/blob/apriltag_cobot_example/src/ac_training_lab/apriltag_hardware_demo/hand_eye_calibration.ipynb), it’s also based on the cobot, but the procedure remains the same.

sgbaird, 2025-07-16:
Thanks! Could you clarify how we get that 4x4 matrix without having some kind of CAD representation or a specific pickup step with physical calibration? (Or maybe I missed that part?)

sgbaird, 2025-07-17:
Daniel Hocevar, who works in SDL6 and has experience with AprilTags mentioned visual servoing, which is in the same spirit as the gradient descent idea.

> The idea is you derive a control law to model how the velocity of the camera (assuming mounted on robot end effector) affects the positioning of your features (in this case april tags) with respect to some sort of target.

Video: https://www.youtube.com/watch?v=HZnH8C4ku3Y&t=5s

I had a look and https://github.com/lagadic/visp seems to be [the most popular on github](https://github.com/search?q=visual+servoing&type=repositories&s=stars&o=desc). Not much that's [popular by way of pure Python approaches](https://github.com/search?q=visual+servoing+language%3APython&type=repositories&l=Python&s=stars&o=desc), though looks like visp has some Python bindings.

ZZZhaoz, 2025-07-17:
> [@ZZZhaoz](https://github.com/ZZZhaoz) also, would the offsets in `apriltag_hardware_demo.ipynb` be something manually measured (e.g., based on a CAD file / assembly)? Or something where you put the arm in the position it's supposed to be in some sense?

There are three kinds of offsets in the notebook:  

1. **Eye-to-hand offset**  

   This offset comes from hand–eye calibration and is applied in `convert()` to calculate the position the end effector should move to.  

2. **`GRIPPER_TOTAL_OFFSET` and `GRIPPER_TOTAL_OFFSET_PLACEMENT`**  

   This offset is obtained by repeatedly adjusting the offset and testing different apriltag positions and poses to find the best grasping accuracy.  

   It includes the deviation between the gripper and the end effector, and may also contain some random error or residual systematic error (likely coming from the error in the hand–eye calibration result itself), which I cannot measure precisely.  On the cobot, I haven’t found a perfect offset to fully compensate for this error.

3. **`z + offset` when descending to grab or place the vial**  

   For example, `z + 20` is the offset in the z-direction between the AprilTag center position and the actual grasping or placement position.  

   For `z + 60`, `z + 70`, `z + 80`, these are offsets used during testing as intermediate positions to prevent the cobot from colliding with the vials or the turntable while moving.

sgbaird, 2025-07-17:
Ah, ok - thank you!

> This offset is obtained by repeatedly adjusting the offset and testing different apriltag positions and poses to find the best grasping accuracy.

I take it this could also be done manually, snapping a picture of the AprilTag of interest, calculating the pose in the end effector / AprilTag frame, recording the pose in the robot base frame, moving the end effector by hand to the ideal location and recording the pose again, and then finding some kind of difference between the three. Does that sound reasonable?

Like you said, the other alternative is to adjust those numbers and try moving to those positions until it goes to the right location (which is visually confirmed by a human).

ZZZhaoz, 2025-07-17:
Positive, I took the second approach since I didn’t find a consistent systematic error, as the deviation varied each time. But it’s still worth a try.

sgbaird, 2025-07-17:
cc @danielhocevar

sgbaird, 2025-07-21:
aside: @ZZZhaoz could you take some of the auto-download fixes in https://colab.research.google.com/drive/18-WF_GoNFaZXC0TZAXvvQPDhPcTlvZFL?usp=sharing and reflect those in your latest version?

ZZZhaoz, 2025-07-22:
Here is the commit: https://github.com/AccelerationConsortium/ac-training-lab/commit/4c27fc55213be448afb0316d2fffae90bc663ebe

sgbaird, 2025-08-25:
@ZZZhaoz could you link (or create) the PR(s) associated with this one?

ZZZhaoz, 2025-08-26:
PR：https://github.com/AccelerationConsortium/ac-training-lab/pull/444