AccelerationConsortium/ac-dev-lab #230: Standalone notebook with an image pre-supplied for AprilTag recognition

sgbaird, 2025-04-17:
And camera calibration

Google colab

sgbaird, 2025-05-14:
cc @ZZZhaoz

ZZZhaoz, 2025-05-15:
I'm planning to use `OpenCV` for camera calibration. For AprilTag detection, the official `apriltag` Python wrapper in [a starting point that uses the offical AprilTag repo](https://colab.research.google.com/drive/1vGwPBQedS8ygqwkh-Z5RpdgXTb5Wngrp#scrollTo=YUQOJXPvz4Y8) doesn't support passing camera parameters for pose estimation. I'll use `pupil-apriltags`, which allows pose estimation by accepting camera intrinsics and tag size as input.

ZZZhaoz, 2025-05-17:
Hi, I’d like to push my code to a new branch, but I currently don’t have push access to this repository. Could you please grant me the permissions?

sgbaird, 2025-05-20:
Added

ZZZhaoz, 2025-05-21:
I’ve completed the work and created a new Colab notebook in the apriltag_demo branch. Please review when you have time.

sgbaird, 2025-05-21:
Thanks! Could you include an open in colab link? I think you should be able to reference to the github file in the branch directly

ZZZhaoz, 2025-05-21:
[image]

sgbaird, 2025-05-21:
Clear structure, logical flow, nice use of colab params, visually appealing / effective use of markdown . Looks great!

Few comments and suggestions:

- use `%pip install ...` rather than the clone and import
For the text copied from the AprilTag README, put in quotes and include link to README
- nice on the scalable SVG helper function
Consider including a link to a read-only Google doc and/or a PDF file with the example AprilTag mentioned (i.e., static / hard-coded resources)
- use colab upload file/files built in functionality for the folder upload
- use numerical parameters instead of strings for the Google colab variable syntax for rows, columns, and size (int, int, float)
- link to resource(s) used for camera calibration, i.e., source link(s). Same idea with quoted text if direct copy
- include hard-coded / static version of the checkerboard images (ideally ones that you take yourself). Display one or two of these images
- hard-code calibrate_camera.py or print the contents of the file to help with transparency. I somewhat lean towards removing the dependency on the repo / need to install and instead just transferring the calibrate Python file to the notebook. I lean away from using cmd line args and instead using the Python function directly 

Have you tested this workflow with your own printed tag/checkerboard? (Not sure if you have easy access to a printer)

ZZZhaoz, 2025-05-21:
Thanks for the detailed feedback! I’ll make those updates later this week.

And yes, I’ve printed and tested both the AprilTag and the checkerboard images myself. Detection and calibration worked as expected, so the workflow has been fully tested on my end

ZZZhaoz, 2025-05-22:
- use `%pip install ...` rather than the clone and import
Done, all dependencies are now in the notebook.
- For the text copied from the AprilTag README, put in quotes and include link to README
Done.
- nice on the scalable SVG helper function
- Consider including a link to a read-only Google doc and/or a PDF file with the example AprilTag mentioned (i.e., static / hard-coded resources)
Done, added a link to a Google Doc.
- use colab upload file/files built in functionality for the folder upload
Done.
- use numerical parameters instead of strings for the Google colab variable syntax for rows, columns, and size (int, int, float)
Done.
- link to resource(s) used for camera calibration, i.e., source link(s). Same idea with quoted text if direct copy
Done. The camera calibration part is based on ChatGPT's explanation; only two short quotes are directly copied from [AprilTag](https://github.com/AprilRobotics/apriltag).
- include hard-coded / static version of the checkerboard images (ideally ones that you take yourself). Display one or two of these images
Done, added two example images and uploaded a zip file to repo.
- hard-code calibrate_camera.py or print the contents of the file to help with transparency. I somewhat lean towards removing the dependency on the repo / need to install and instead just transferring the calibrate Python file to the notebook. I lean away from using cmd line args and instead using the Python function directly 
Done, calibrate_camera.py and tag_to_svg.py are now fully embedded and used as functions.

sgbaird, 2025-05-22:
Awesome, excited to take a look! Looks like the colab link from above may go to the old version. Is there a new link?

sgbaird, 2025-05-22:
Sorry, realized it's the same link (see edit above). Looks really good! I'll test it out tomorrow hopefully and provide any final edits. Do you mind opening a pull request?

sgbaird, 2025-05-22:
Added a few more minor comments and suggestions at https://github.com/AccelerationConsortium/ac-training-lab/pull/255#issuecomment-2902808109

closed 2025-05-25 by sgbaird

ZZZhaoz, 2025-06-03:
Apriltags used on vials [apriltag_sheet_vial.pdf](https://github.com/user-attachments/files/20576690/apriltag_sheet.32.pdf)