AccelerationConsortium/ac-dev-lab #231: Prototyping the 8-hr auto-restart mechanism and stream restoration

sgbaird, 2025-04-28:
Related:
- https://github.com/AccelerationConsortium/ac-training-lab/issues/202 (focused on getting the API set up and pulling stream URL automatically)
- https://github.com/AccelerationConsortium/ac-training-lab/issues/212
- https://github.com/AccelerationConsortium/ac-training-lab/issues/223

Based on discussion with @Neil-YL:

How should the following tasks be handled (e.g., same manager, different managers):
1. Close the current stream and start a new one every 8 hours
2. Log the recent status of each stream into a CSV every 10-15mins
3. Report the latest download link to a database

Or is it better to separate these into different scripts or services?

Maybe the monitor (task 2) as a separate service.  Maybe the auto-restart and download links as a schedule-based service (e.g., scheduled task with Prefect) to trigger the workflow: first retrieve the download link for the latest completed video (not the currently active stream), then close the current broadcast and create a new one.

Another idea is to have a monitor create new stream when `device_name` is not in active stream list, and an 8-hr schedule to end all active streams

> 8-hr schedule to end all active streams
This one could also restart the stream, to minimize downtime (i.e., loop through each stream one at a time --> [end stream, restart stream]), but maybe this would cause issues if the monitor and this both try to restart the same stream.

It could be one code/service to both end/restart and monitor, when executing ending and restart, try to move out of the loop of the checking and when the restart completed back to the monitor loop.

---

There also needs to be a service that auto-downloads the videos and post-processes them (high bandwidth, so probably don't want to mix with the scripts that ensure uptime). This might be long-running processes on the order of 1+ hrs for each video, where GPUs would significantly accelerate the process. If there are only a few streams, a single service might be able to handle this. If there are 50+ streams, then there would need to be multiple machines doing the processing. This might justify some kind of trigger-based mechanism.. (instead of having a few dedicated machines doing the processing).

Neil-YL, 2025-04-30:
Sum up for the test yesterday:

I set a 5 hours restart yesterday at around 16:35.

So it will trigger end and restart at 21:35, 2:35, 7:35, we should have 3 video for each device, and a live now. 
1. 16:35-21:35
2. 21:35-2:35
3. 2:35-7:35
4. 7:35-now

Now we only has the footage for:

- OT2: 1. 4hrs 56mins; 2. 32mins; 3. no footage; 4. no live;
- A1mini: 1. 4rs56mins; 2. no footage; 3: 1hr52mins; 4. on live;
- Pi4-test: 1. no footage; 2. 32mins; 3. no footage; 4. no live;
 
The 32 mins records are very likely be triggered by the 2am reboot on the zero2w, but not sure why A1mini didn't have a similar 32mins video.

[image]

sgbaird, 2025-05-01:
As a side note, I think it may be worth changing the stream latency to "low-latency", and possibly adding a delay (unless adding a delay doesn't help). There seems to be missing segments (e.g., when the OT-2 was removed from the office for the event), as noted in a separate issue (I think the OT-2 equipment monitoring issue or the original Zero 2W issue).
[cid:d31037e8-050a-4f89-aa50-c024665cd509]

Neil-YL, 2025-05-01:
> As a side note, I think it may be worth changing the stream latency to "low-latency", and possibly adding a delay (unless adding a delay doesn't help). There seems to be missing segments (e.g., when [...]

I think it is because it happened to have a no data issue during the removal period. I will change the latency to low in next start.

Neil-YL, 2025-05-01:
[image]

[image]

Add enableMonitorStream: false to the broadcast creating codes as suggested in one stackoverflow, seems not helping improving the uptime 
14:00 - 19:00: lose around 30mins
19:00 - 24:00: lose around 1hrs 8/15mins
0:00 - 5:00: lose around 1hrs 40mins and 2 hrs
5:00 - now: never start until I reboot zero2w

I will think about the cross devices end and restart.
Basic scheme:
Manager: End device.py on zero2w --> End YTB broadcast --> Restart YTB broadcast --> Restart device.py on zero2w
(It also could be:  End YTB broadcast --> Restart YTB broadcast --> End device.py on zero2w --> Restart device.py on zero2w?)
Monitor: Find which device broadcast is down  --> End device.py on zero2w -->  Restart YTB broadcast --> Restart device.py on zero2w

sgbaird, 2025-05-01:
This sounds great! I thing this leverages the scheduling strengths of cron pretty well, and like you said avoids complicating device top high which already took a while to get it to be robust.

sgbaird, 2025-05-01:
> Per our conversation, I think the decision was to include something in the title (maybe original video id? Or could be timestamp) that we can keep the same between the original and the processed videos, and add some kind of note in the title like: [processed, 16x]. @Neil-YL

Source: https://github.com/AccelerationConsortium/ac-training-lab/issues/223#issuecomment-2845867241

sgbaird, 2025-05-01:
@Neil-YL instructions https://github.com/ACC-HelloWorld/5-data-logging#aws-lambda-function

Neil-YL, 2025-05-05:
I finished the setup on AWS Lambda. I will start working on the device.py and the crontab restart.

[image]

Neil-YL, 2025-05-06:
OK I think I have finished the deployment and I put it on the Bambu A1mini stream zero2w for test. I set a reboot at 05:00, 13:00 and 21:00 with crontab. 
Now the playlist has 27 video so it would be 30 tomorrow. 
[image]

If everything works as expected I will create a PR later.

sgbaird, 2025-05-07:
Awesome! Funnily, just as I was reading this, I got a yt notification (9:01 pm)

sgbaird, 2025-05-07:
This is amazing! 45 s downtime is not bad at all either 🚀 thank you!

I'll try to figure out how we can get the lambda function centralized. Do you mind pasting a snippet of the lambda function code here?

The 8 hour restart mechanism looks solid!

Neil-YL, 2025-05-07:
> I'll try to figure out how we can get the lambda function centralized. Do you mind pasting a snippet of the lambda function code here?

https://colab.research.google.com/drive/1ApwqbTkNmUfOAlsTlrTavd4yR_tprOPT?usp=sharing

Colab to generate the deployment.zip for Lambda.

sgbaird, 2025-05-08:
Also, just to confirm - is this UTC time? We probably talked about this already

https://colab.research.google.com/drive/1ApwqbTkNmUfOAlsTlrTavd4yR_tprOPT?usp=sharing

sgbaird, 2025-05-08:
Looks like something odd happened with a couple last night.

https://colab.research.google.com/drive/1ApwqbTkNmUfOAlsTlrTavd4yR_tprOPT?usp=sharing

Neil-YL, 2025-05-08:
> Also, just to confirm - is this UTC time? We probably talked about this already

Yes.

> Looks like something odd happened with a couple last night.

I was working on the PR yesterday, and the streaming Zero2W was very laggy, so I stopped the service and paused the stream for a while. I restarted it around 2 PM, so the footage is incomplete for these two clips.  (5am to 1pm and 1pm to 9pm).

sgbaird, 2025-05-08:
Ah, got it. Thanks!

Neil-YL, 2025-05-08:
> Ah, got it. Thanks!

Can I merge this one https://github.com/AccelerationConsortium/ac-training-lab/pull/241 so that I can deploy the code on other devices.

sgbaird, 2025-05-08:
Difficult to share Lambda function directly on AWS, so instead long-term we may switch to having the lambda function on GitHub and automatically deploy to cloud via [AWS Chalice](https://aws.github.io/chalice) for example, per comment in:

https://stackoverflow.com/questions/58441717/how-to-share-lambda-function-with-another-user-in-organization

Neil-YL, 2025-05-09:
[image]

sgbaird, 2025-05-09:
Amazing!

sgbaird, 2025-05-09:
Another thought about naming schemes: we could have the hostnames and video titles be non-workflow-specific (e.g., SDL5 camera-7t3j) and have the playlists be workflow/hardware in the playlist titles. This could make it more flexible as labs evolve and needs change.

Neil-YL, 2025-05-09:
> Another thought about naming schemes: we could have the hostnames and video titles be non-workflow-specific (e.g., SDL5 camera-7t3j) and have the playlists be workflow/hardware in the playlist titles. This could make it more flexible as labs evolve and needs change.

I would prefer to have same/similar name for the broadcast/video and playlist since the` start stream and add to playlist` function is using the same parameter to match the streamkey, create broadcast title and add to playlist. Or we have to add another mapping between the two (within the YouTube API utils script). 

Using a non-workflow-specific host name seems very reasonable to me though.

sgbaird, 2025-05-09:
Those points make sense to me!

sgbaird, 2025-05-09:
- Try to change stream key to one-time stream key (otherwise, can 
- Separate device name into camera name + workflow name
- Remove stream key from device secrets, return stream key from lambda
- Lambda should receive both camera name and workflow name
- Camera hostname without workflow / hardware name (e.g., no "ot2")

Aside:

Maybe worth noting there is the ability to ["update"](https://developers.google.com/youtube/v3/live/code_samples) the title and description of a stream, though single use stream keys would be a bit more secure since lambda would be returning the stream key (and leaking a single stream key causes issues with all future videos with that stream, and could be problematic especially for private streams).
[image]

Neil-YL, 2025-05-12:
Now all three livestream devices are on new script with the workflow_name and cam_name, using returned one-time stream key from YouTube API for streaming.

Hostname also changed on Zero2w.

sgbaird, 2025-05-21:
Looks like [all streams are down](https://studio.youtube.com/channel/UCHBzCfYpGwoqygH9YNh9A6g/videos/live?filter=%5B%5D&sort=%7B%22columnType%22%3A%22date%22%2C%22sortOrder%22%3A%22DESCENDING%22%7D). Maybe something to do with the lambda function?

There are many repeats of ones that are scheduled, which could also mean the device restart mechanism is having some issues (e.g., lambda function returns error after successfully scheduling a stream, then device script restarts due to error, triggering lambda function again, but only up to X number of times due to systemd restart limit?).

Wondering if related to credentials/authorization and possible expired token.

[image]

Maybe worth moving to using 

> Difficult to share Lambda function directly on AWS, so instead long-term we may switch to having the lambda function on GitHub and automatically deploy to cloud via [AWS [...]

The difficulty of sharing the lambda function is a bit annoying. Maybe worth creating a separate set of credentials that owns the lambda function and that both of us share. Likewise, could be good to move to Chalice.. thoughts?

Neil-YL, 2025-05-21:
Checked the log on the device service:

> 
> Sending to Lambda: {'action': 'end', 'cam_name': 'cam-fb7p', 'workflow_name': 'OT2-LCM-TrainingLab', 'privacy_status': 'private'}
> Status code: 200
> Response text: OT2-LCM-TrainingLab ended successfully
> Lambda 'end' succeeded: OT2-LCM-TrainingLab ended successfully
> Sending to Lambda: {'action': 'create', 'cam_name': 'cam-fb7p', 'workflow_name': 'OT2-LCM-TrainingLab', 'privacy_status': 'public'}
> Status code: 500
> Response text: Error during 'create' for device 'OT2-LCM-TrainingLab': <HttpError 500 when requesting https://youtube.googleapis.com/youtube/v3/playlistItems?part=snippet&alt=json returned "Internal error encountered.". Details: "[{'message': 'Internal error encountered.', 'domain': 'global', 'reason': 'backendError'}]">
> Traceback (most recent call last):
>   File "/home/ac/ac-training-lab/src/ac_training_lab/picam/device.py", line 109, in call_lambda
>     response.raise_for_status()
>   File "/usr/lib/python3/dist-packages/requests/models.py", line 1021, in raise_for_status
>     raise HTTPError(http_error_msg, response=self)
> requests.exceptions.HTTPError: 500 Server Error: Internal Server Error for url: https://u7xxb347xucah6slu7fjs5ei3q0kjnar.lambda-url.us-east-2.on.aws/
> During handling of the above exception, another exception occurred:
> Traceback (most recent call last):
>   File "/home/ac/ac-training-lab/src/ac_training_lab/picam/device.py", line 134, in <module>
>     raw_body = call_lambda(
>                ^^^^^^^^^^^^
>   File "/home/ac/ac-training-lab/src/ac_training_lab/picam/device.py", line 124, in call_lambda
>     raise RuntimeError(f"HTTP error occurred: {e} - Response: {response.text}")
> RuntimeError: HTTP error occurred: 500 Server Error: Internal Server Error for url: https://u7xxb347xucah6slu7fjs5ei3q0kjnar.lambda-url.us-east-2.on.aws/ - Response: Error during 'create' for device 'OT2-LCM-TrainingLab': <HttpError 500 when requesting https://youtube.googleapis.com/youtube/v3/playlistItems?part=snippet&alt=json returned "Internal error encountered.". Details: "[{'message': 'Internal error encountered.', 'domain': 'global', 'reason': 'backendError'}]"> 

Unsure why it did create a stream but when listing the playlist it returned an error to the device.py so it cannot get the stream key from the create function.

I will look into that.

Neil-YL, 2025-05-21:
> Checked the log on the device service:
> 
> > Sending to Lambda: {'action': 'end', 'cam_name': 'cam-fb7p', 'workflow_name': 'OT2-LCM-TrainingLab', 'privacy_status': 'private'}
> > Status code: 200
> > Response text: OT2-LCM-TrainingLab ended successfully
> > Lambda 'end' succeeded: OT2-LCM-TrainingLab ended successfully
> > Sending to Lambda: {'action': 'create', 'cam_name': 'cam-fb7p', 'workflow_name': 'OT2-LCM-TrainingLab', 'privacy_status': 'public'}
> > Status code: 500
> > Response text: Error during 'create' for device 'OT2-LCM-TrainingLab': <HttpError 500 when requesting https://youtube.googleapis.com/youtube/v3/playlistItems?part=snippet&alt=json returned "Internal error encountered.". Details: "[{'message': 'Internal error encountered.', 'domain': 'global', 'reason': 'backendError'}]">
> > Traceback (most recent call last):
> > File "/home/ac/ac-training-lab/src/ac_training_lab/picam/device.py", line 109, in call_lambda
> > response.raise_for_status()
> > File "/usr/lib/python3/dist-packages/requests/models.py", line 1021, in raise_for_status
> > raise HTTPError(http_error_msg, response=self)
> > requests.exceptions.HTTPError: 500 Server Error: Internal Server Error for url: https://u7xxb347xucah6slu7fjs5ei3q0kjnar.lambda-url.us-east-2.on.aws/
> > During handling of the above exception, another exception occurred:
> > Traceback (most recent call last):
> > File "/home/ac/ac-training-lab/src/ac_training_lab/picam/device.py", line 134, in 
> > raw_body = call_lambda(
> > ^^^^^^^^^^^^
> > File "/home/ac/ac-training-lab/src/ac_training_lab/picam/device.py", line 124, in call_lambda
> > raise RuntimeError(f"HTTP error occurred: {e} - Response: {response.text}")
> > RuntimeError: HTTP error occurred: 500 Server Error: Internal Server Error for url: https://u7xxb347xucah6slu7fjs5ei3q0kjnar.lambda-url.us-east-2.on.aws/ - Response: Error during 'create' for device 'OT2-LCM-TrainingLab': <HttpError 500 when requesting https://youtube.googleapis.com/youtube/v3/playlistItems?part=snippet&alt=json returned "Internal error encountered.". Details: "[{'message': 'Internal error encountered.', 'domain': 'global', 'reason': 'backendError'}]">
> 
> Unsure why it did create a stream but when listing the playlist it returned an error to the device.py so it cannot get the stream key from the create function.
> 
> I will look into that.

Seems too frequent request of the API due to the service on the device?
https://github.com/googleapis/google-api-python-client/issues/2013#issuecomment-1363336814

Neil-YL, 2025-05-21:
Weird that even though it returned 500 error to the device.py, it still added the new stream to the right playlist

[image]

Neil-YL, 2025-05-21:
I will reboot all other devices.

Neil-YL, 2025-05-22:
> (aside: looks like the lambda function URL is in the logs you posted - maybe worth refreshing the URL or adding some mild authentication)
> 
> Get Outlook for Android<https://aka.ms/AAb9ysg>
> […](#)

👌

Neil-YL, 2025-05-27:
> Difficult to share Lambda function directly on AWS, so instead long-term we may switch to having the lambda function on GitHub and automatically deploy to cloud via [AWS [...]

If I understand correctly:

- Create a new repo for the Lambda function
- Edit Lambda scripts on this repo
- Set up AWS account credentials on GH secrets
- Commit to trigger GH action using Chalice to deploy new scripts on Lambda(?)

The only thing I need help would be the AWS credentials? I think I need an IAM user credentials within the AC organization?

sgbaird, 2025-05-27:
I'm ok with either option - pursue Chalice now (the steps you described are what I was thinking of too), or create a shared account like what was mentioned in that issue. I'll double check on AWS credentials. Not sure if I gave you admin permissions.

Neil-YL, 2025-05-27:
I thought the IAM user credentials is something other than our account? like a IAM user for Lambda deployment.

Neil-YL, 2025-05-27:
Not sure if the IAM user also needs the permit to visit S3 (for the pickle token) or there is a better practice. I will look into this while preparing the other setup for the shared managed Lambda.

sgbaird, 2025-05-28:
Seems more complicated than I realized. Created [a new issue](https://github.com/AccelerationConsortium/ac-training-lab/issues/271) specific to Chalice.

sgbaird, 2025-05-31:
The 8 hour restart seems to be working quite well. Frequently, the stream only has a minute of downtime for the 8 hour chunks.