AccelerationConsortium/ac-dev-lab #243: Set up and install equipment monitoring cameras to SDL5

sgbaird, 2025-05-08:
Four planned. Possible naming scheme:

Device name (i.e., vendor name), research purpose, location

E.g., "OT-2 Liquid Color Matching AC Training Lab"

High throughput dissolution apparatus (HTDA), hot melt (HM), critical micelle concentration (CMC)

Long name for title (keep below 100 characters for full title), short name for hostname.

@ZeqingBao

ZeqingBao, 2025-05-09:
Thanks!

1. OT-2 high throughput dissolution apparatus SDL5 (HTDA)
2. OT-2 hot melt SDL5 (HM)
3. OT-Flex surfactant critical micelle concentration SDL5 (CMC)
4. OT-2 repeatability and reproducibility SDL5 (RR)

Neil-YL, 2025-05-12:
I've finished most of preparation work for the other three monitoring cameras. 

Followings are the names I used:
1. OT2-HTDA-SDL5
2. OT2-HM-SDL5 (This is the one already in SDL5 lab)
3. OTFlex-CMC-SDL5 
4. OT2-RR-SDL5 

I will ask @sgbaird to change the network on the devcies to "UofT" network, and I think they would be ready to be brought to your office.

sgbaird, 2025-05-29:
Seems to be working well. One camera coming back to us for now. Can re-open if needed

closed 2025-05-29 by sgbaird