AccelerationConsortium/ac-dev-lab #248: Set up OT-2 simulators on fresh Raspberry Pi 3B+'s

sgbaird, 2025-05-09:
Found https://github.com/Opentrons/opentrons-integration-tools/tree/main/raspberry-pi via https://github.com/Opentrons/opentrons-integration-tools/issues/13

Ordered 6 3B+'s and SD cards / power cables.

Neil-YL, 2025-05-13:
Our OT2 version:

> **Robot Server Version**
> v8.2.0

Neil-YL, 2025-05-13:
Our simulator is up on tailscale: 

[image]

[image]

@sgbaird Feel free to play with.

Neil-YL, 2025-06-12:
[image]

If connected to a monitor, the simulating system won't boot.
(and might break the autostart of the Tailscale)