AccelerationConsortium/ac-dev-lab #250: Create Prefect MWE between two colab notebooks

sgbaird, 2025-05-13:
Including some basic setup instructions (links and a few bullets are fine), and focusing on converting a particular MQTT script into a Prefect script.

sgbaird, 2025-05-23:
Colab notebooks:

- [OT-2 simulator](https://colab.research.google.com/drive/1RYD1br7WeH8tIQM439p2XeeGQmUqZ_BO?usp=sharing) (basic usage, including what a somewhat verbose MQTT setup looks like)
- [OT-2 device](https://colab.research.google.com/drive/1BVFvDe53EHcqqXEecrUqT2teV5UXzv5J?usp=sharing) (Prefect)
- [OT-2 orchestrator](https://colab.research.google.com/drive/14kX4jaeywtGXMQygMJ5eQzLD-EYjDdJM?usp=sharing) (Prefect)

JANERUBBISHTOEAT, 2025-05-27:
Just a few questions to clear:

- What is the python eversion now? If we have a condo environment, I could use the yaml file to set up.
- Is it ok to hardcode API keys (username / passwords) in the file and upload to GitHub? I see some of the files in this repo doing that.
- I will use the hardcoded API key for MQTT, and will have create myself a API key for Prefect (since I don’t have them in the `-prefect` files). Is that correct?

Thanks.
Zheyuan

sgbaird, 2025-05-27:
> Just a few questions to clear:
> 
> - What is the python version now? If we have a conda environment, I could use the yaml file to set up.

Anywhere between 3.10 and 3.12. See setup.cfg in top level dir

> - Is it ok to hardcode API keys (username / passwords) in the file and upload to GitHub? I see some of the files in this repo doing that.

Case by case. The public MQTT broker is the main one that's public, but otherwise most should be kept private (GitHub secrets, my_secrets.py files that are .gitignored)

> - I will use the hardcoded API key for MQTT, and will have create myself a API key for Prefect (since I don’t have them in the `-prefect` files). Is that correct?

Sounds good, the sgbaird one is fine for public demos. Yes, you'll need to authenticate on Prefect privately. I think you can use Jupyter notebook's / colab 's password input (passkey?) dialogue box

> Thanks.
> Zheyuan
>

sgbaird, 2025-05-27:
Forgot to mention: https://ac-training-lab.readthedocs.io/en/latest/contributing.html (you can use the environment.yml file for environment). Some of this is boilerplate instructions, but useful for things like pre-commit and should be mostly accurate

JANERUBBISHTOEAT, 2025-05-28:
New question:
Prefect server side is working well except:

1. `p300.pick_up_tip(tiprack_2["A2"])` is been complained by `opentrons` with:

   - `opentrons_shared_data.errors.exceptions.InvalidStoredData: Error 4008 INVALID_STORED_DATA (InvalidStoredData): This labware cannot be used wells_covered_dense because it is less dense than an SBS 96 standard`
     it's likely a problem with either `opentrons` doing unnecessary safety checks or the description JSON file for `tiprack_2` indeed does not meet requirements.

2. We want to run a server that monitor clients request, for that we normally do

    ```python
    main_flow.serve(
        name="my-first-deployment",
        tags=["onboarding"],
        parameters={"goodbye": True},
        interval=60,
    )
    ```

    but
    1. we are in a Jupyter notebook which has a `asyncio` already (to enable all the async operations we make to the notebook, e.g., running code cells)
    2. we are starting a new `asyncio` by `main_flow.serve()`. This is not allowed. We will get Exception `RuntimeError: This event loop is already running`

    Luckily, the solution with least efforts would be **running the code blocks out of a Jupyter notebook**, but **in a ordinary python file**. But in that case we lose all feature from the notebook, such as cell debugging, quick partial re-run... and also cannot be posted on colab.

Would you like to proceed with my suggestions to fix the issues above?

For 1 I will replace `.pick_up_tip` with `.move_to` to **avoid error** and to proceed with the code execution.
For 2 I will **move the code to a separate python file** and run it from the terminal, so that we can use `main_flow.serve()` without issues.

sgbaird, 2025-05-28:
@JANERUBBISHTOEAT thanks for the update! Could you share a small Colab reproducer for (1)? This would help a bit with debugging.

For (2), it would be really nice to have it on Colab (very much from a demo/showcasing/tutorial perspective). Could you include a small Colab reproducer for (2), also? (If you're using a Prefect API key, please share privately over email so I can run). We can also think about using `.deploy` instead of `.serve` or running a worker in a subprocess, but I think it would be good to see the reproducer first to poke around a bit.

sgbaird, 2025-05-29:
Thanks for committing one of the notebooks. Here's the open in colab badge:

[image]

(Easy to generate automatically with https://openincolab.com/)

JANERUBBISHTOEAT, 2025-05-29:
Here attaches screenshots to prior issues.

## Issue 1

[image]

[image]

[image]

## Issue 2

[image]

sgbaird, 2025-05-29:
- `%pip install prefect opentrons` on colab leads to a much lower version of Opentrons being installed, which may be the reason for the pick up tip function causing issues. As a workaround, `%pip install opentrons` followed by `%pip install prefect` seems to install things OK, despite some warnings. Alternatively, `%pip install opentrons prefect` should work, but a note or a version spec on opentrons should be added. Not sure if this is colab specific
- the Opentrons simulation package seems to allow you to specify a path to a labware JSON file that isn't present on the filesystem (i.e., doesn't throw an error). You can either use wget for the JSON files (since these files won't change frequently and are already on `main`) or use `load_labware` directly and have it point to the non-existent files. Latter would keep things simpler for now, which I think is better. Yanghuang can correct me if there's a specific reason to keep using `load_labware_from_definition` rather than `load_labware` pointing to the relative path in practice
- Collapse the init() function back into the script. We can add it back later, but I think it will keep things simpler (generally, this init code would only ever run once when the machine boots up). Ideally I think we want to get rid of the `global` usage
- can you prepare a second notebook that's just a prefect serve?

sgbaird, 2025-05-29:
Thanks for clarifying! So I guess use `load_labware_from_definition` still?

Neil-YL, 2025-05-29:
I think so! Or perhaps we can define the labware json file in the block above the protocol. (Not sure this would work or not, iirc there is a documentation says that only json file generated by Opentrons’ labware designer would work in the protocol.

JANERUBBISHTOEAT, 2025-05-29:
Yes. Method `load_labware_from_definition` is wat I kept same from the sample notebook.

Thanks for the hints, here's some results:

- Install packages one by one solved the install failure issue 👍
- Yes, JSON file was one of the reasons that I developed on local. Jupyter notebook on Colab always work alone, cannot access other files in repo. I can definitely wget/clone the repo then use JSON in notebook, and will do in future commits.
- Will Collapse the `init()`
- By notebook with only `prefect` serve, do you mean one like the sample notebook [OT-2 Orchestrator - Prefect](https://colab.research.google.com/drive/14kX4jaeywtGXMQygMJ5eQzLD-EYjDdJM?usp=sharing) / OR do you mean a notebook that only contains a prefect `main_flow.serve()` call without any code from `opentrons`🤔

Also I'm not sure how the two sample notebooks would connect to each other, I understand that the `_device_(mqtt/prefect)` notebook is a **server** for device control/simulation and exposes the `main_flow` to be used at **client** side, while the `_orchestrator_` notebook seems quite empty rn, with only a few `opentrons` sample codes.

sgbaird, 2025-05-30:
Cross-communication via prefect cloud (I think I sent an invite to organization)

Yes, the thought was for debugging and troubleshooting, something that only depends on Prefect and runs e.g., their hello world example, also communicating via prefect. The incompatibilities and behavior of opentrons package could be the source of some of the asyncio issues

sgbaird, 2025-05-30:
Great! Your point about init makes a lot of sense. Might be good to switch back to that. Do you mind opening a discussion on Prefect GitHub asking about this situation and include the link to it here? (In particular this idea of hardware initialization steps, that should typically only ever run once at startup of a device, but where some info needs to be shared - hence explicit use of globals or putting it at top-level of script)

Thanks for prepping the second notebook. Good to see it's isolated to Colab (ipykernel as I think you mentioned) and Prefect. Maybe worth trying with `nest-asyncio` and this simple example. Up to you

sgbaird, 2025-05-30:
Thanks! Not ideal,  but could try running in subprocess. You can use %%writefile magic too so that the code is still inline

JANERUBBISHTOEAT, 2025-05-31:
Sure I can do that!
Will update a link here once I post the discussion.
I'll also test different hacks and see which one would our possible neat workaround.

Neil-YL, 2025-05-31:
Not sure if I missed anything, but I thought there should be a similar approach for our HITL setup on the LCM Huggingface space. One Colab notebook should have an active worker running (workpool-worker->deployment-flow-task), so that the other Colab can communicate with it and trigger the script.

sgbaird, 2025-05-31:
I probably should have mentioned that. Thanks for bringing up! @JANERUBBISHTOEAT some relevant places in Yanghuang's code:
 https://huggingface.co/spaces/AccelerationConsortium/OT-2-LCM/blob/main/prefect_utils.py

Notice that gets called here https://huggingface.co/spaces/AccelerationConsortium/OT-2-LCM/blob/main/app.py#L64

https://huggingface.co/spaces/AccelerationConsortium/OT-2-LCM/blob/main/maintenance_flow.py

If I'm not mistaken, https://huggingface.co/spaces/AccelerationConsortium/OT-2-LCM/blob/main/DB_utils.py#L85 is what gets triggered eventually via `app.py` within `find_unused_wells` function from DB_utils.py, and start_prefect_worker just needs to have happened prior to that being called (?).

Neil-YL, 2025-06-01:
To give a clear explanation:

There is a deployment on our Prefect Cloud named `wells-maintenance`, with the entry point located in `maintenance_flow.py `within our Hugging Face Space, with the flow `request_wells_maintenance`. We also have a work pool on Prefect Cloud called OT2-pool, and a worker assigned to this pool is triggered when our LCM Hugging Face Space is built.

JANERUBBISHTOEAT, 2025-06-01:
Very clear explanation.
Just double check my understanding:

**flow**: a function. plain python code
**deployment**: a configuration of a function. can be created by `flow_name.deploy()` or `prefect deploy --...`
**work pool**: a middleware to store & manage deployments
**worker**: a terminal machine connected to OT-2 that runs deployments from designated work pool

```mermaid
graph TD
    Flow["Flow<br/>(Python function)"]
    Deployment["Deployment<br/>(Configuration of Flow)"]
    WorkPool["Work Pool<br/>(Manages Deployments)"]

    subgraph "Remote Machine<br/>(e.g. OT-2 connected system)"
        Worker["Worker<br/>(Executes Deployments)"]
    end

    Flow --> |".deploy()/prefect deploy"| Deployment
    Deployment --> |"--work-pool ot2-pool"| WorkPool
    WorkPool --> |pulls| Worker
    Worker --> |"prefect worker start"| WorkPool
```

JANERUBBISHTOEAT, 2025-06-01:
By the arrow direction I intended to show the development process:

- First write the function (flow)
- Only then, I can use commands/methods to push the deployment to Prefect cloud
- To specify a work-pool by the `--work-pool` args
- Only then, use `prefect worker start` to start a worker that pulls the deployment from the work-pool and executes it

I'll be around the downtown campus on Monday and Tuesday for some errands, but I’m not sure yet if I'll have time to drop by the AC office.
What’s the best way to reach you in case I’m free?

Neil-YL, 2025-06-01:
> By the arrow direction I intended to show the development process:
>
> * First write the function (flow)
> * Only then, I can use commands/methods to push the deployment to Prefect cloud
> * To specify a [...]

I think so! Maybe some details vary with your configuration.

> I'll be around the downtown campus on Monday and Tuesday for some errands, but I’m not sure yet if I'll have time to drop by the AC office. What’s the best way to reach you in case I’m free?

I would be at the office from Wednesday hopefully, and then I will be working in person everyday during work hours.

JANERUBBISHTOEAT, 2025-06-10:
> > Will update a link here once I post the discussion.
> 
> Any update here?

I looked up [`ipykernel`](https://github.com/ipython/ipykernel) as it is the direct problem we were facing with `asyncio`.

---

> ## Bad news
> Yes, I was also looking into `nest-asyncio`. It looked hopeful when it ran without `asyncio` errors: there was no error, no warning, no complaints, and also no deployments.
> ...
> My guess: still `asyncio` stuff. Jupyter or kernel does have certain limitations, and Colab as a Jupyter hosting service, has even more limitations (e.g., can't change python version).

The `nest-asynio` didn't work as expected. Explained in: <https://github.com/ipython/ipykernel/issues/1381>

---

> Do you also mind opening one up in relation to this Colab /ipykernel / asyncio issue?

`ipykernel` repo does not have a discussion page. I looked up existing issues, where I find the discussions on requesting the ability to use `asyncio` in Jupyter has been confirmed not feasible for the reason below:

> asyncio is always running in an IPython kernel.

Mentioned in both open & closed issues.
 https://github.com/ipython/ipykernel/issues/548
https://github.com/ipython/ipykernel/issues/372

---

## Questions:

- Since the approach used by @Neil-YL has already been verified to work, would it be advisable for me to follow the same method when setting up the workflow?
- If so, would it be possible to arrange a meeting with @Neil-YL to go over the deployment procedure? I noticed that part of the deployment seems to be handled through the terminal and may not be fully captured in the current files.
- Just to confirm, if we proceed with the same approach as @Neil-YL's examples, would the main objective in this issue be "to have a Prefect version of color mixing service up"?

Neil-YL, 2025-06-10:
I’ll be in the office during regular hours every day until June 20. Let me know when you’d like to go over the deployment procedure.

I believe you can use a similar approach to mine, though there will be some differences—since in my case, all the code runs within the same Hugging Face Space, whereas you’ll need two Colabs to interact.

sgbaird, 2025-07-04:
@JANERUBBISHTOEAT any updates?

ZZZhaoz, 2025-07-17:
@sgbaird I’ve read through the previous discussion and would like to clarify a few points:

Is it acceptable if we only use a simulated device (without considering any future connection to real hardware)? Since we are only using a standalone notebook, based on earlier discussions, it seems this wouldn’t work directly with `.serve()` inside the notebook.

If we only use `.deploy()`, it can run remotely on Prefect Cloud (simulation only) and remains real communication between notebooks. However, if we need to eventually connect to real OT-2 hardware, it seems we would need to follow @Neil-YL’s approach (using a worker and `.serve()` on a machine that has access to the hardware).

Could you confirm if this understanding is correct or if I’ve missed anything?

ZZZhaoz, 2025-07-18:
Referring to @Neil-YL's implementation and the current project setup, I asked GPT, and it gave the following. Does this look correct?

```
┌──────────────┐   trigger    ┌────────────────┐   schedule    ┌──────────────────┐
│ Orchestrator │ ───────────▶ │ Prefect Cloud  │ ─────────────▶ │ HF Space Worker  │
│ (Colab/HF)   │              │ Deployment API │                │ device_flow.serve│
└──────────────┘              └────────────────┘                └──────────────────┘
```
- A worker runs inside the HF Space, connected to a specific work pool, and stays online waiting for tasks.

- The Colab/HF orchestrator triggers a Prefect Cloud deployment via client.create_flow_run().

- Prefect Cloud receives the trigger and assigns the flow run to the corresponding work pool.

- The worker in HF Space picks up the task and executes device_flow.py (either in simulation mode or connected to the real OT-2 in the future).

sgbaird, 2025-07-18:
For the tutorial of this issue, hugging face won't play into it (or at least probably shouldn't). For the specific implementation of the color matching demo on hugging face spaces, this contains the relevant code: https://github.com/AccelerationConsortium/ac-training-lab/issues/250#issuecomment-2925253705.

Have you looked at the four notebooks in https://github.com/AccelerationConsortium/ac-training-lab/issues/250#issue-3060507695?

sgbaird, 2025-07-18:
I'll try to clarify more about the specifics of the color matching workflow later. For now maybe good to focus on writing two scripts, orchestrator.py (run_deployment) and device.py (serve) in separate local kernels, with both kernels logged into Prefect Cloud (you can use your personal free-tier cloud workspace for the demo). Once you do, a screen recording of the communication happening would be great.

ZZZhaoz, 2025-07-18:
https://github.com/user-attachments/assets/f0a23ec0-cfa1-48de-a6c2-189d740d758f

sgbaird, 2025-07-18:
Awesome. That's exactly it. Could you make a PR with those scripts? Maybe put them within a new `prefect` directory within https://github.com/AccelerationConsortium/ac-training-lab/tree/main/src/ac_training_lab/ot-2/_scripts

closed 2025-08-15 by sgbaird