AccelerationConsortium/ac-dev-lab #254: Explore and reproduce MADSci

sgbaird, 2025-05-21:
Here's a starting point for a workflow orchestration platform from Argonne National Laboratory, Rapid Prototyping Laboratory: https://github.com/AD-SDL/MADSci_Examples.

The idea would be to look through the notebooks and try to walk through/reproduce the examples where possible.

sgbaird, 2025-05-21:
Cc @angelasun-20

angelasun20, 2025-05-22:
I went through the node_notebook and experiment_notebook. The node notebook taught me that a NodeDefinition of a Node instance allows us to differentiate between a Node instance and the Node module. I think that the NodeInfo is very helpful to see the values of each of the attributes of a node. As well, I found that the error messages printed when an argument is not given or the number of joint angles is not 4 are insightful in explaining to the user why an error occurred. Something else that I learned was that time.sleep() can be used to pause the program while it is running so that we can wait for the node to be ready for unlocking or resetting. 

In the experiment notebook, I found that each experiment has a unique ID, which makes it easier to identify which experiment is currently being run. As well, I noticed that a stack or queue is used to push the resources to a location.

sgbaird, 2025-05-22:
Thanks! Also, I think you mentioned it worked on Google colab. Did you need to make any modifications to the notebook or did it work as-is?

angelasun20, 2025-05-22:
The node notebook worked, but for the experiment notebook, there was an error that no module 'madsci' was found.

sgbaird, 2025-05-22:
Thanks! For the experiment notebook, could you open up an issue on the corresponding repo and provide a link to that issue here?

angelasun20, 2025-05-22:
https://github.com/AD-SDL/MADSci_Examples/issues/1

sgbaird, 2025-05-31:
Could you follow up with them and try to explore a bit getting a MADSci Lab instance running? (There might be some existing places in the code or docs you can look to)