AccelerationConsortium/ac-dev-lab #257: MWE for windows single board computer

sgbaird, 2025-05-22:
@yakavetsiv has zima boards that he's been using rather than RPi's 

Doesn't overheat, able to install full Linux or Windows, x86 (not arm)

He's tried out a number of SBCs

[image] ~300 CAD

[image] ~800 CAD

[image] [image], SDL5 also using. ~1k CAD (no windows)

Related to OT-2 simulator RPi's issue

sgbaird, 2025-05-22:
Ilya lent a Zima board. @linx5o do you have a link to Windows installation instructions you used?

linx5o, 2025-05-22:
@sgbaird 
No real guide online. I will try to make one today and put it in GH.

Neil-YL, 2025-06-11:
Hi @linx5o Since Zimaboard only has a built-in storage of 32GB, I am wondering if you are using it with an external storage (SSD or HDD)

Neil-YL, 2025-06-11:
> Aside: [@Neil-YL](https://github.com/Neil-YL) was able to get the activation key through UofT.

Actually I skipped the product key in installation process and run some codes as instructed on the UofT Microsoft authentication page with CMD, once the device was able to connect to UofT network.

linx5o, 2025-06-11:
@sgbaird @Neil-YL 
You can also add storage after. Kingston, crucial and Samsung are some well known brands. Look for 2.5 inch SSDs.

There should be a SATA cable included in the box with the zima board. Power off before plunging and unplugging. SATA is not hot plug safe.

The drive may also need to be partitioned/formatted in disk manager.

sgbaird, 2025-06-12:
Thanks! Appreciate the hints. Wouldn't have thought too much about the hot plug safety.

Considering https://a.co/d/iCiZt0P (based on Amazon search for "2 5 inch ssd 128gb"). 64 GB or 128 GB would probably be fine, but this 240 GB one is pretty highly rated and direct from crucial store. Seem ok?

linx5o, 2025-06-12:
@sgbaird That SSD looks good.

linx5o, 2025-07-04:
Since we are using windows, I believe that any generic usb wifi dongle of amazon should work. 

[image]

sgbaird, 2025-08-21:
Bought several of https://www.amazon.ca/TP-Link-Archer-T3U-Adapter-Wireless/dp/B07P6N2TZH/, which Nipun has been using.