AccelerationConsortium/ac-dev-lab #259: Explore and reproduce ALabOS

sgbaird, 2025-05-24:
Starting point: https://cedergrouphub.github.io/alabos/. From here, it seems best to tackle it sequentially - read to the bottom of the homepage, click next at the bottom-right, etc.

Note: When asked to set up MongoDB, set up a free-tier MongoDB Atlas cluster (rather than doing a local installation).

Ideally, keeping track of general notes and impressions about the experience, and opening up [issues on the ALabOS repo](https://github.com/CederGroupHub/alabos/issues) when necessary.

angelasun20, 2025-05-24:
Yes, for sure!

angelasun20, 2025-05-26:
Hi, I was wondering if I need to use Atlas CLI or Atlas UI and if I should use my U of T email when I register for an account.

sgbaird, 2025-05-27:
@angelasun20 thanks for the update! Could you open an issue on the ALabOS repository and also include the link to that issue here?

angelasun20, 2025-05-28:
I finished going through the tutorial, and the issue was fixed thanks to Yuxing's help.

sgbaird, 2025-05-28:
It looks like they have a simulation mode as well as support for mock devices and tasks

https://cedergrouphub.github.io/alabos/start_lab.html

https://cedergrouphub.github.io/alabos/device_definition.html#mocking-the-device

sgbaird, 2025-05-28:
Could you open a new issue on ALabOS related to that and add the link here again? If you could include a screen recording starting from scratch that might help speed things up.

sgbaird, 2025-05-31:
Note: if you're following the installation instructions for installing from source code (https://alab-management.readthedocs.io/en/latest/installation.html#install-from-source-code), it will install into the Python environment you're running it in (ideally a virtual environment). This should be done rather than trying to copy the source code folder around.

You might consider trying this out in Colab if you're having trouble running Python on your local machine.

angelasun20, 2025-05-31:
I tried it in Colab, but I am still getting a module not found error. I saw in the terminal that the editable project location for alab_management is incorrect, but I was wondering how to change it. 

[image]

angelasun20, 2025-05-31:
https://colab.research.google.com/drive/132cna7W5RgMoOV0CbgTLONZaTCZgEsWD?usp=sharing

angelasun-20, 2025-06-02:
I fixed the module not found error, but there is now a ServerSelectionTimeoutError, even though I already changed the configuration file based on the MongoDB Atlas configuration.

angelasun-20, 2025-06-03:
Here is an issue where I opened suggested changes that could be made to the tutorial for future users so that they can better understand: https://github.com/CederGroupHub/alabos/issues/93