AccelerationConsortium/ac-dev-lab #260: Explore and reproduce various lab orchestration platforms

sgbaird, 2025-05-24:
Especially those listed in https://github.com/AccelerationConsortium/awesome-self-driving-labs#workflow-orchestration

sgbaird, 2025-06-25:
Possibly ROS2, NODE-RED, and Prefect (even though we've done a fair bit with Prefect, would be good to get Angela's perspective).

sgbaird, 2025-07-10:
Hi @angelasun-20 , could you give a high-level update about the current status of things when you get a chance?

angelasun-20, 2025-07-10:
Here is a recording of the first flow for Node-RED that I created following the document instructions: 

https://github.com/user-attachments/assets/0d9c6cbe-232e-4e98-ab20-f2c24cd81cda

I think we could implement a flow for a self-driving lab using an inject node first and then an HTTP request node to read data from a device and use the function node with code from Python that would determine the next step depending on the data received from the previous node. I also think that a CSV node could be used for remembering the data, and a debug node could be used to output a message, as shown in the recording.