AccelerationConsortium/ac-dev-lab #263: Explore and reproduce Bluesky

sgbaird, 2025-05-27:
https://blueskyproject.io/bluesky/main/index.html, which on the main page links to a tutorial as a starting point

angelasun20, 2025-05-31:
Yes, I can try this. The link in the email that you sent says that the invitation is not found, so I was unable to accept it.

angelasun-20, 2025-06-08:
https://github.com/NSLS-II/Bug-Reports/issues/208

angelasun-20, 2025-06-10:
I finished going through the Bluesky tutorial and I was able to reproduce the plots that were shown.

sgbaird, 2025-06-11:
Thanks! I noticed you closed the other issue. Could you leave a comment there clarifying how you solved it?

Could you create a screen recording (narrated would be nice, but not required) showing the setup from scratch up to producing the plots?