AccelerationConsortium/ac-dev-lab #324: Create Zero 2W-based temperature/pressure/humidity/gas sensor minimal working example

sgbaird, 2025-06-12:
@Neil-YL thoughts? I'm considering having it be based on Zero 2W so that it could be integrated directly as an additional service with the picam setups, and it could be added to the same Zero 2W as the picam (possibly with a slight modification to the 3D printed mount).

In terms of which sensor and the code: https://github.com/AccelerationConsortium/ac-training-lab/issues/13

We've done this with MicroPython code, but maybe worth switching to CircuitPython and using adafruit-blinka library or similar.

Neil-YL, 2025-06-12:
Do we have one on hand? Being curious about the pins/connector it uses.

If it’s sensitive to temperature, we might need to consider the heat generated by the Pi when designing the packaging.

Also, what general data interface would you suggest if we plan to integrate it into an additional service? (Like, the Picam has YouTube streaming as its output.)

sgbaird, 2025-06-12:
Had trouble finding it. Just checked with Luthira to see if he knows where it is.

Probably to MongoDB Atlas, for now. Maybe with blinka we could use pymongo and still access the sensor data. Eventually, this could also happen within the same script as the streaming (maybe with a `DISABLE_SENSOR` toggle in case someone wants to ignore that part and just implement the camera). I could also see us adding that data as part of the clock text overlay (see issue). We'd need to check with one of the cameras to see if it can handle the extra load and maintain the stream.

We might still want to have the MWE be completely separate from the camera, and leave at least one zero 2w as a standalone, dedicated device. However, after that we may embed it into the picam script with that disable flag since we wouldn't be able to easily add the data to the text overlay if it were running as a separate "service". Gets messy trying to get two independent scripts to interact with the same file in realtime.

Aside: I think MongoDB has a time-series style collection

sgbaird, 2025-06-13:
Wow, nice finds!

Zero 2WH isn't on pishop.ca, but  it's on digikey.ca which is great - https://www.digikey.ca/en/products/detail/raspberry-pi/SC0721/24627135

I've ordered from The PiHut before I think. Thanks! EDIT: just ordered 7 to UofT. Planning to order 2 to UBC for Daniel (along with a few extra peripherals I forgot to send him with).

I'll also order some more [BME680 sensors](https://www.digikey.ca/en/products/detail/adafruit-industries-llc/3660/7915571).

I think we were running low on M2.5?

Could you create and share a DigiKey cart link here with what we need?

Starting point for the M2.5 hardware:
- https://www.digikey.ca/en/products/filter/screws-bolts/572?s=N4IgjCBcoOwAxVAYygMwIYBsDOBTANCAPZQDaIAzABxgAsYMIAuoQA4AuUIAyuwE4BLAHYBzEAF9xhAExkQAW2kA6AKzNxQA
- https://www.digikey.ca/en/products/result?s=N4IgTCBcDaILZgHQFYAEA7ArgFxAXQF8g

Just realized we can also get some micro-usb extension cables (so we don't have to order entirely new power supplies). Probably 10 of https://www.digikey.ca/en/products/detail/startech-com/USBUBEXT50CM/22160418.

The form factor of the ones we have can be a bit problematic when trying to put lots on the same surge protector.

Neil-YL, 2025-06-13:
m2.5 screws and standoff: https://www.digikey.ca/en/products/detail/adafruit-industries-llc/3299/6596885

Some long USB cables might be helpful as well (in SDL2 we had to rearrange other devices with the power bar given the length of the cable) https://www.digikey.ca/en/products/detail/cvilux-usa/DH-20M50056/13177301

sgbaird, 2025-06-18:
Nice! All looks good to me. The form factor of that power adapter certainly seems more compact. Agreed about micro-usb extensions being kind of niche/expensive. I think this will provide more flexibility. At some point, if there are several cameras in close proximity, a USB splitter adapter could be good too, but probably a down-the-line thing.

kelvinchow23, 2025-06-18:
above 2amps, i found it hard to find a compact power adapter (maybe 2.5a was still compact). im not sure what the actual power draw of  the camera setup is, it looks its 1.2A per pi zero2w. wouldnt hurt to try to run two off this power adapter, i'll keep this in mind in the near future