AccelerationConsortium/ac-dev-lab #329: Fixturing AprilTags to the various pieces of equipment / samples

sgbaird, 2025-06-13:


zweaung1014, 2025-07-07:
In our latest tests, we put the camera 14.2 cm away from the apriltag, which has an outer-box size of 23 mm and an inner-box size (tagsize_mm) of 13 mm. 

When we entered 13 mm for tagsize_mm, right here,
[image]
the "Distance from camera" measured by the code is 8.3 cm, which is way different from the actual distance (14.2 cm).

When we entered 23 mm for tagsize_mm, the "Distance from camera" measured by the code is 14.69 cm, which is pretty close to actual distance. 

Since you mentioned that tagsize_mm is referring to interior box, this is a little weird. I don't think we can confirm that tagsize_mm is supposed to be the outer-box size yet. Maybe we just got lucky.

zweaung1014, 2025-07-07:
[image]

This is the photo we were working with.

sgbaird, 2025-07-07:
Ah, I see. @ZZZhaoz could you comment here? I remember we had some discussion related to this.

sgbaird, 2025-07-07:
@zweaung1014 could you also copy the camera intrinsics here?

zweaung1014, 2025-07-08:
@ZZZhaoz Yes, we performed camera calibration, but we used these pre-supplied ones in the ZIP file:
[image]
I'm not sure if these photos were taken using the same camera. From what you're saying, it's probably the calibration then. We will redo calibration manually without using the ones from the ZIP file.

zweaung1014, 2025-07-09:
We redid the camera calibration again today with 6x9 squares (which means 5 rows x 8 columns in terms of inner corners), of size 23 mm, and tried to run detection again. There seems to be a slight offset of about 0.858 cm in the measured 'Distance from camera', but it's pretty consistent. I organized the results in this file: [https://utoronto-my.sharepoint.com/:x:/g/personal/larry_aung_mail_utoronto_ca/EZtRSqvMeXFDgmTI3fbeh9wBV3tugZblfOM0FfROVuVyUQ?e=rLAqQh](url)

If the offset is consistent, we can probably account for it and use it. What do you guys think?

sgbaird, 2025-07-09:
Nice update! Thanks for putting the spreadsheet together. The consistent offset may be related to what @ZZZhaoz shared just barely in https://github.com/AccelerationConsortium/ac-training-lab/issues/229#issuecomment-3048717444 (probably should have mentioned sooner, my bad). Copying a snippet below:

> There was a consistent issue with how the convert() function handled the rotation matrix from hand-eye calibration. Previously, when only the translation vector was applied (ignoring rotation), there was always a 1–2 cm error. I’ve now revised the transformation logic to properly include the rotation matrix, and based on what I can see from the livestream, the grasping seems accurate.
> ...
> Run my notebook on [this branch](https://github.com/AccelerationConsortium/ac-training-lab/tree/apriltag_cobot_example) and check if the grasping accuracy looks good. For now, please focus only on grasp precision. I'll make refinements later.

See https://github.com/AccelerationConsortium/ac-training-lab/blob/apriltag_cobot_example/src/ac_training_lab/apriltag_hardware_demo/apriltag_hardware_demo.ipynb and https://github.com/AccelerationConsortium/ac-training-lab/commit/72e71f5d84c2737179ab67e5e09c69004c01de3c for the fix. I'm trying out that notebook. If all goes well, then I think we'll update the calibration tutorial with the new functionality to avoid the error.