AccelerationConsortium/ac-dev-lab #330: Integrate AprilTag detection with the UR3e Python control

sgbaird, 2025-06-13:
Including doing the camera calibration with the checkerboard.

@xiaomguo could you point me to where the Python control files are being stored or to the person managing this? (public is great, but no worries if it's to a private repo as long as I have access).

xiaomguo, 2025-06-17:
I’ve pushed the Python control files to the repo and added you as a collaborator, so you should have access now.
It’s currently a private repo, and I haven’t set the .gitignore yet. I’d appreciate it if you could help with that.
Here is the link: https://github.com/xiaomguo/sdl2_solid_dosing

sgbaird, 2025-06-28:
Haven't tested out the AprilTags code on windows, though I thought maybe @ZZZhaoz had run the AprilTags demo code locally.