AccelerationConsortium/ac-dev-lab #336: Potential redesign of vial holders

sgbaird, 2025-06-16:
@xiaomguo, have a look at the video from https://github.com/AccelerationConsortium/ac-training-lab/issues/15#issuecomment-2724260933. Using a flexure-based design can help relax the positioning tolerance for vials and similar labware. Doesn't necessarily have to mimic the one shown, but figured you might be interested / might be worth mentioning.

xiaomguo, 2025-06-17:
Thanks! I have a few ideas now. Sacrificing one vial position for an AprilTag sounds reasonable. Btw,  do you happen to know what size AprilTags we could use?

Right now the vials are uncapped, but ideally they should have caps. We need to add a decap function before placing them on the balance. I haven’t seen your decapper design before — if it seems suitable, I’d love to try integrating it.

sgbaird, 2025-06-17:
See https://youtube.com/shorts/jQLtg0luPNc, but it still needs to be automated via RS485 and the student who was working on the RS485 programmatic control is no longer at the AC: https://github.com/AccelerationConsortium/ac-training-lab/issues/123#issue-2686286647. Technically, the simplest is to push the two buttons with the robotiq gripper..

This is the one we have, it works reasonable well, but not sure I'd get it again - https://ca.robotshop.com/products/dh-robotics-dh-robotics-automated-screw-cap-decapper-intelligent-capping-machine (and depending on the workflow might be overkill). Lmk what you think. More than happy to see this put to better use

xiaomguo, 2025-06-24:
I’ve redesigned our vial holder based on the flexure-style idea. It's easier to place vials. The apriltag plate is 25.5*25.5mm. Happy to hear any thoughts on this!

[image]

[image]

sgbaird, 2025-06-24:
Awesome! Would you be willing to post a video showing you manually putting a vial in one of the slots? (Mimicking with your hand what the robotic arm might be doing, especially if there are some minor position deviations). You can attach videos under 100 MB to GitHub directly, or if larger than that, you could upload to YouTube as unlisted and share the link here.

xiaomguo, 2025-06-24:
https://github.com/user-attachments/assets/597c8154-1025-4b55-bb3c-9b1478d7cad5

sgbaird, 2025-06-27:
@xiaomguo not urgent, but could you also add a video of you putting the vial into the original vial holder, doing a bit of the same stress testing?