AccelerationConsortium/ac-dev-lab #351: Set up T16-P electronic controls

sgbaird, 2025-07-02:
LAC powered by 12V, connected to T16-P and Pi Zero 2W

Pi Zero 2W powered separately for now (until we get a step-down regulator and shunt regulator)

Should be able to control position and read position. See Kelvin's repo

sgbaird, 2025-07-03:
@chasedamagician could you share a picture and a video?

chasedamagician, 2025-07-07:
I have finished setting up the Raspberry Pi Zero as well as the Actuonix LAC with the linear actuator and I am able to control the actuator using the Raspberry Pi with no issues. However I am unsure on how I would be able to receive data from the LAC, the only output pins I could find on the datasheet are in the USB port which I cannot use due to the limited available USB ports on the board.

[image]

I will try to investigate if it's possible to use UART with the RX/TX GPIO pins on the Raspberry Pi, but I am not sure that is possible with the LAC.

Additionally I have finished making a prototype for the stationary compression plate for the UTM. Next I will start designing the plate that will be attached to the carriage of the actuator. 

[image]

Here is a video of the LAC and the Raspberry Pi running a basic program:

https://github.com/user-attachments/assets/d30973d4-2009-4eb9-a251-11c8812c7567