AccelerationConsortium/ac-dev-lab #353: Flashing RPi os lite onto zero 2W

sgbaird, 2025-07-02:
And setting up software for controlling the position and reading out the position data

sgbaird, 2025-07-03:
We'll also set this up with tailscale and UofT network.

Tailscale: https://github.com/AccelerationConsortium/ac-training-lab/discussions/183#discussioncomment-12464477

Connecting to UofT: https://github.com/AccelerationConsortium/ac-training-lab/discussions/183#discussioncomment-12502531

For UofT network, I'll provide the login. For tailscale, you'll download tailscale on your personal computer, sign in with GitHub, and then select the Acceleration Consortium tailnet. Lmk when you get to that point.

When you set up RPi OS Lite (https://www.raspberrypi.com/software/operating-systems/), you'll need to use the mobile hotspot in the office. I'll share that over slack.

sgbaird, 2025-07-14:
@chasedamagician do you have a step-down voltage regulator? Can you power the ADC separately for now? Check with Kelvin about a 5V power supply with bare wire leads (you may need to doing some wire stripping).

chasedamagician, 2025-07-17:
I have completed the setup with the ADC and Raspberry Pi and can now read voltages ranging from 0 to 3.3 V corresponding to the actuator's position along the track. I will begin working on the next stages of the 3D design and start researching methods to measure force output from the linear actuator.

sgbaird, 2025-07-17:
Nice! Could you include a video demo and copy the snippet of the test code that you're running here?

chasedamagician, 2025-07-17:
Code: 
`import pigpio
import time
import sys
import board
import busio
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn

i2c = busio.I2C(board.SCL,board.SDA)
ads = ADS.ADS1115(i2c)
chan = AnalogIn(ads, ADS.P0)
print("Voltage:", chan.voltage)   
pi = pigpio.pi()
GPIO = 18

print("I2C devices found", [hex(i) for i in i2c.scan()])

ads1115 = 0x48

if not ads1115 in i2c.scan():
    print("Could not find ads1115")
    sys.exit

pi.set_servo_pulsewidth(GPIO,1000)
def move_actuator(pos):
    pi.set_PWM_dutycycle(GPIO, pos)
    print(pos)
    time.sleep(1)
def read_position():
    print("Voltage:", chan.voltage)
    print ("Raw ADC Value:", chan.value)
    position = 300-(chan.voltage * (300/3.3))
    print("Position:", position, "mm")

for i in range(0,255,10):
    move_actuator(i)
    read_position()
for i in range(0,255,10):
    move_actuator(255-i)
    read_position()
    
move_actuator(0)
   
`

chasedamagician, 2025-07-17:
'''import pigpio
import time
import sys
import board
import busio
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn

i2c = busio.I2C(board.SCL,board.SDA)
ads = ADS.ADS1115(i2c)
chan = AnalogIn(ads, ADS.P0)
print("Voltage:", chan.voltage)   
pi = pigpio.pi()
GPIO = 18

print("I2C devices found", [hex(i) for i in i2c.scan()])

ads1115 = 0x48

if not ads1115 in i2c.scan():
    print("Could not find ads1115")
    sys.exit

pi.set_servo_pulsewidth(GPIO,1000)
def move_actuator(pos):
    pi.set_PWM_dutycycle(GPIO, pos)
    print(pos)
    time.sleep(1)
def read_position():
    print("Voltage:", chan.voltage)
    print ("Raw ADC Value:", chan.value)
    position = 300-(chan.voltage * (300/3.3))
    print("Position:", position, "mm")

for i in range(0,255,10):
    move_actuator(i)
    read_position()
for i in range(0,255,10):
    move_actuator(255-i)
    read_position()
    
move_actuator(0)
   
'''

chasedamagician, 2025-07-17:
Code: ```import pigpio
import time
import sys
import board
import busio
import adafruit_ads1x15.ads1115 as ADS
from adafruit_ads1x15.analog_in import AnalogIn

i2c = busio.I2C(board.SCL,board.SDA)
ads = ADS.ADS1115(i2c)
chan = AnalogIn(ads, ADS.P0)
print("Voltage:", chan.voltage)   
pi = pigpio.pi()
GPIO = 18

print("I2C devices found", [hex(i) for i in i2c.scan()])

ads1115 = 0x48

if not ads1115 in i2c.scan():
    print("Could not find ads1115")
    sys.exit

pi.set_servo_pulsewidth(GPIO,1000)
def move_actuator(pos):
    pi.set_PWM_dutycycle(GPIO, pos)
    print(pos)
    time.sleep(1)
def read_position():
    print("Voltage:", chan.voltage)
    print ("Raw ADC Value:", chan.value)
    position = 300-(chan.voltage * (300/3.3))
    print("Position:", position, "mm")

for i in range(0,255,10):
    move_actuator(i)
    read_position()
for i in range(0,255,10):
    move_actuator(255-i)
    read_position()
    
move_actuator(0)
   
```

sgbaird, 2025-07-28:
Ordered two of these sensors