AccelerationConsortium/ac-dev-lab #364: Multi-fidelity Branin (branin_fidelity) optimization competition MWE

sgbaird, 2025-07-04:
> In a fourth competition, you can create a multi-fidelity benchmark that uses a cost budget rather than an iteration budget. People are allowed to strategically choose which fidelity they evaluate at for each of their iterations, and the stopping criteria is when their "running cost" exceeds the allowed budget.

Aside: users should be able to calculate cost without consuming their function evaluation budget

sgbaird, 2025-07-04:
> Also, here's a multi-fidelity example with the running cost I mentioned. I haven't tested it in a while, but it may help get the idea across
> 
> https://github.com/sparks-baird/self-driving-lab-demo/blob/main/notebooks/6.2.1-multi-fidelity-continuous.ipynb ([via docs](https://self-driving-lab-demo.readthedocs.io/en/latest/tutorials.html)]
> 
> [image]

_copied from https://github.com/AccelerationConsortium/ac-training-lab/issues/262#issuecomment-2964808840_

Amanichab, 2025-07-31:
For the multi-fidelity setup, which fidelities should we include? Perhaps something a bit more realistic?
Also, in general, should we provide some data or not?
Regarding the evaluation metric, should we keep the same one used in the Vanilla BO competition — where the score is the average of the values obtained at the 40th iteration, assuming the same budget is maintained?