AccelerationConsortium/ac-dev-lab #374: Purchase and test UPS for vertical lift module

sgbaird, 2025-07-04:
@yakavetsiv do you have the link for where you ordered the following? Would you recommend it? 

I noticed it's on [amazon](https://www.amazon.ca/APC-Smart-UPS-SmartConnect-Uninterruptible-SMC1500C/dp/B077Y62GSJ/) and [directly from seller](https://www.apc.com/ca/en/product/SMT1500C/apc-smartups-line-interactive-1500va-tower-120v-8x-nema-515r-outlets-smartconnect-port+smartslot-avr-lcd/).

I also came across https://www.digikey.ca/en/products/detail/eaton-tripp-lite/SMART1500LCDT/4439114

[image]

Curious if you had any other tips or recommendations for choosing the UPS (specs, brand, etc.)

pranshu-ui, 2025-09-15:
Forgot to update on this, we have received the package. Should I take it to the NRC when I head down there?

sgbaird, 2025-09-15:
Yes, sounds good. You'll also want to bring one of the openflexure microscopes, too

closed 2025-09-30 by pranshu-ui