AccelerationConsortium/ac-dev-lab #382: BO / Prefect HiTL slack integration tutorial

sgbaird, 2025-07-07:
This assumes that the entire campaign runs from start to finish with no changes, where the evaluation is human-in-the-loop via the Prefect slack integration. If the script errors out or fails for whatever reason, the campaign is effectively ruined (addressed in a separate issue). See https://github.com/AccelerationConsortium/ac-training-lab/pull/344#issuecomment-3036955249 and the parent issue.

sgbaird, 2025-09-18:
https://github.com/AccelerationConsortium/ac-dev-lab/pull/344#issuecomment-3308636052

Daniel0813, 2025-09-18:
> Based on conversation with @Daniel0813:
> 
> The overall deliverable / success metric for this sub-issue / minimal working example is a video screen recording showing:
> 
> * User runs a Python script that starts a BO campaign via Ax
> * Ax suggests a new experiment, which triggers a Prefect-based slack message (human-in-the-loop)
> * The user goes to slack, looks at the suggested experiment, and "runs the experiment" (for the sake of the demo, using https://huggingface.co/spaces/AccelerationConsortium/branin, similar to [this video tutorial](https://youtu.be/4tnaL9ts6CQ?si=SeGxZk1ZvK-yF9W6))
> * The user clicks on the link in the slack message to get the Prefect UI and clicks "resume run", followed by entering the objective function value from the previous step
> * The Python script receives the input back, resumes, and passes the control back to Ax
> * Ax suggests a new experiment, and this loop is carried out 4-5 times (obviously if too many, that would be a long, boring video, but 4-5 times probably helps get the point across)

follow the instructions in copilot-instructions.md closely. Use context7 and download the documentation for Ax 0.5.0 and Prefect 3. Refer to this frequently during your development. Create the minimal working example for this issue. Refer to the examples in ac-dev-lab\scripts\prefect_scripts