AccelerationConsortium/ac-dev-lab #403: Set up pipettin bot

sgbaird, 2025-07-09:


naikymen, 2025-07-15:
Hello!

Have you joined the Discord server yet? Here is an invite link for you: https://discord.gg/Rz4PSFsJ

There's a thread for your build journey there, next to the other two builders, Naruki and Yifang.

I can answer questions and help out over there. Please ping me when you join :)

sgbaird, 2025-07-17:
Just joined! I need to find someone who can build it on our side. Will keep you posted

angelasun-20, 2025-07-28:
Hi, I don't have access to join the Discord after I pressed the link to join.

sgbaird, 2025-07-28:
@angelasun-20 here's a direct link to the docs (I already joined the discord server from before): https://stepwisedocs.com/docs/projects

[image]

angelasun-20, 2025-07-31:
@naikymen, when I was following the instructions to build the structure of the pipettin bot from here: https://www.stepwisedocs.com/docs/projects/pipettin-bot-mk3-4/guides/structure-13, I found that instead of two 2020P-400 profiles that are needed to slide onto the vertical profiles, I have two other profiles with large screws inside them, which are shown in the photo below. 

[image]

Also, here is a photo showing all of the rods that I have:

[image]

angelasun-20, 2025-08-11:
Hi, when attaching the baseplate to the structure following here: https://www.stepwisedocs.com/docs/projects/pipettin-bot-mk3-4/guides/baseplate-12, it says that we need M4-button head screws, but they were not provided in the box, so I was wondering what size the screws need to be.

naikymen, 2025-08-11:
Hi @angelasun-20 nice to meet you. Here is another link: https://discord.gg/NbeJPdyn (expires in a week).

Please join and we'll chat over there.

Are these the "screws" you mentioned? These are pulleys for the Y-axis, folks must have left them installed for you.

[image]

sgbaird, 2025-08-26:
Just a note that it seems we'll need to 3D print some parts, based on a conversation with Angela

angelasun-20, 2025-09-04:
Here is the status of the assembly of the pipettin bot so far:
- The aluminum structure is completed, but some of the rods are not completely aligned straight
- The baseplate is not yet screwed to the aluminum structure to stabilize it, but it is in the correct place
- The motion system has been completed fully
- The back panel is attached using screws
- The tool-changer remote, head plate, and connection have already been completed
- The x and z-axis motor cables are already attached to the correct location, but the y-axis one is not yet
- The Raspberry Pi and Duet 2 boards are installed
- The mini display is installed and connected
- The software and firmware portion of the assembly guide here: https://www.stepwisedocs.com/docs/projects/pipettin-bot-mk3-4/guides/software-and-firmware-44, and all of the portions below it have not been completed yet

naikymen, 2025-10-22:
Hi everyone! How is the build coming along?

sgbaird, 2025-11-06:
@naikymen any chance you could provide a drop-in STL or STEP file for the pipettin bot? Just need something to add into a lab layout for layout planning. Noticed the freecad files, but would be great if you could send something I can import to Fusion 360 directly 🙏 

- https://gitlab.com/open-la/pipettin-bot/-/tree/master/models?ref_type=heads
- https://gitlab.com/open-la/pipettin-bot/-/blob/master/models/MK3/ASSEMBLY.FCStd
- https://gitlab.com/open-la/pipettin-bot/-/tree/master/models/general

angelasun-20, 2025-11-11:
Just to give an update of the status of the assembly:
- everything has been completed except for the network setup of the Raspberry Pi 
- some of the motor cable wires (endstops) are not connected yet