AccelerationConsortium/ac-dev-lab #406: Explore and reproduce NODE-RED

sgbaird, 2025-07-10:
Angela, could you also link to any tutorials or resources that you found helpful?

angelasun-20, 2025-07-10:
I used this page to download and run Node-RED on my laptop: https://nodered.org/docs/getting-started/local.

This page has tutorials that I followed, specifically the first and second flows: https://nodered.org/docs/tutorials/.

angelasun-20, 2025-07-10:
Here is a recording of the second flow:
https://github.com/user-attachments/assets/d9bda1e1-588e-4435-b726-c08ffadbce0b

angelasun-20, 2025-07-10:
Yes, Node-RED can be integrated with HiveMQ cloud brokers using MQTT in or MQTT out nodes that can be dragged into the flow.  Examples of this shown below are an inject node leading to an MQTT out node for publishing messages and an MQTT in node leading to a debug node for subscribing to messages.

https://github.com/user-attachments/assets/ed845d7b-994a-49cd-95a8-7701064a9297