AccelerationConsortium/ac-dev-lab #445: Networking in training lab at NRC

sgbaird, 2025-08-26:
The mobile hotspot has some reliability issues, and the upload bandwidth isn't enough to support our long-term plans, so Farhad is checking with the IT team to see what options are available for us getting a separate Internet-enabled network installed, preferably without having to rewire everything from scratch.

pranshu-ui, 2025-09-04:
Looking into BELL plans closest to NRC. 
The closest one is at: 5100 Erin Mills Pkwy Level 1, Mississauga, ON L5M 4Z5
And their contact is: [+19058209648].
Called them and they said to reach out to another competitor as their plan there only reaches up to 0.5mbps, which is essentially unusable. Will reach out to Rogers next.

sgbaird, 2025-09-08:
If it were closer to 100 Mbps, would you expect the price to go down significantly? (Not sure if that came up, no worries if not)

pranshu-ui, 2025-09-08:
Here's Bell's proposal based on the speeds discussed for future reference.

[UOT  BID service proposal.pdf](https://github.com/user-attachments/files/22213774/UOT.BID.service.proposal.pdf)

pranshu-ui, 2025-09-10:
Chatted with a Rogers associate, their plan is covered at the NRC and is faster for less of a cost. Check it out!
https://www.rogers.com/business/internet-services/5G-business-internet

sgbaird, 2025-09-26:
Temporarily, let's put the mobile hotspot near a window in an adjacent lab while we're getting guest access sorted out.

sgbaird, 2025-10-24:
Thanks! It sounds like it uses a walled garden approach. We'll have to check if we can stay logged in on a Pi for an extended time. If not, while it helps with our personal computers (decent speeds), it's not a general-purpose solution we can use. If extended connection works with Pi's and other devices, then maybe we can get extended 6-month access for devices and stick with that.

pranshu-ui, 2025-10-28:
Extended connection on my laptop didn't work- I had to log in again. However, I have the MyCobot arm connected to the same network and that has stayed connected over the weekend entirely.

I think if the session stays on, aka if the device isn't turned off, it will stay connected. Basically there's no automatic reconnect like most networks.

sgbaird, 2025-11-03:
Thanks for the update! This will be especially problematic for the streaming cameras, which have a built-in 8 hr restart. In other words, likely unusable. It may be worth trying to reboot the MyCobot arm to verify that it will stay disconnected.

sgbaird, 2025-12-11:
There are three routers. Is that the case for all three?

sgbaird, 2025-12-15:
Sounds like the two white NOKIA routers (tall cylinders) got moved back to AC by Rachel. The Cudy Routers are separate from this - I'll make a separate issue.

sgbaird, 2025-12-17:
We need to swap out the WiFi credentials for the Pico W that's onboard the wireless color sensor tool within the liquid color-matching OT-2. I'll send you the credentials over WiFi.

[image]

I ran @gurpaul-kochhar through a few things here. @gurpaul-kochhar would you be able to either hop on a call or visit in person to help with this?