AccelerationConsortium/ac-dev-lab #448: pH probe testing (atlas scientific)

sgbaird, 2025-08-28:
[image]

[image]

[image]

[image]

https://atlas-scientific.com/kits/ph-kit/ (I think)

Probably can find some cases where people have used Python. Lmk if you have trouble, at least one comes to mind in the SDL space, but I'd need to dig up.

sgbaird, 2025-08-29:
Nice! Thanks for including the link. Also, for reference, Pico W's have two UART ports. Raspberry Pi single board computers (SBCs) like the Zero 2W, 4B, and 5 have UART as well.

Also came across https://atlas-scientific.com/ezo-accessories/i2c-toggler/ based on this tutorial: https://www.instructables.com/UART-AND-I2C-MODE-SWITCHING-FOR-ATLAS-SCIENTIFIC-E/

Likewise, I found a Python library for i2c communication with Atlas devices, and pH seems to be supported: https://github.com/timboring/atlas_i2c

Both Pico W and Pi SBCs support i2c. See https://robu.in/raspberry-pi-zero-2w-how-to-enable-i2c/ for a related guide with Pi Zero 2W.

Could be treated as a standalone "internet of laboratory things" device in this sense.

owen-melville, 2025-08-29:
Hmm ok. I currently have an Elegoo Uno R3 and a ESP32 WROOM.

owen-melville, 2025-08-29:
Is there a reason to use I2C instead of UART? I don't understand that intrinsically

sgbaird, 2025-08-29:
The micropython perspective bit here resonates with me - https://claude.ai/share/0b3f471f-dbc5-4944-846a-89dabaf4124c. The last bit with a specific example less so. However, I've done a lot with using existing i2c packages and a little with uart, so I'm already a bit biased there.