AccelerationConsortium/ac-dev-lab #457: PiCam for SEM

sgbaird, 2025-09-18:
Please send the credentials to me via slack and pin the message. I might ask you to rename the hostname. Please also include 

You will also need to get this PiCam set up on tailscale if you haven't already. Please login with your GitHub account on tailscale.com and then ping me on slack so I can approve your account. You can follow the instructions on the training lab homepage for how to connect to tailscale.

We have a bunch of mounts in the office and the 3D printed parts. I assume it's just the software that's set up at this point?

Once it's ready and installed, please include a picture of the final setup.

pranshu-ui, 2025-09-19:
The physical setup is ready, will send credentials and software setup details later.

[image]

sgbaird, 2025-09-19:
Nice! We also have some shorter cables (I think in the room with the 3D printer in a medium cardboard box). For the really short cable which is nice and compact, you can flip the pi zero 2w around to get it close to the camera, and then flip the camera vertically (there's a software toggle in my_secrets.py that allows you to flip the camera vertically)

pranshu-ui, 2025-11-03:
I haven't installed tailscale on this camera yet because of the earlier firmware issues. I will set that up tomorrow and send you the creds. Currently though they are still on the Laptop hotspot- but it isn't live becuase I was using the hotspot to set up the arm and SEM actuator. 
I could definitely connect it to the lab hotspot too.

sgbaird, 2025-11-14:
The camera appears to be damaged. Needs replacement

EDIT: some very small pins may have just come loose. Can try again later.

sgbaird, 2025-11-14:
Also, the screws used were too big and I ended up sawing them off and using pliers to get them out. Were the same sizes as specified in the docs used?

sgbaird, 2025-11-15:
New camera worked. Old camera doesn't:

> Codec AVOption preset (Encoding preset) has not been used for any stream. The most likely reason is either wrong type (e.g. a video option with no video streams) or that it is a private option of some decoder which was not actually used for any stream.

Tossing the old camera.